OSVVM/OSVVM #4: Workaround for "tee" keyword needed

cmarqu, 2016-01-29:
It seems that Cadence Incisive 14.2 does not support the VHDL-2008 keyword "tee" (in https://github.com/JimLewis/OSVVM/blob/master/TranscriptPkg.vhd#L172), even with the "-v200x -extv200x" options set. Could you add a workaround please?

JimLewis, 2016-02-10:
I am working toward preparing a dev_cadence branch.  I will be adding Cadence specific changes there.  

If it is an OSVVM bug, I will be fixing that on the main branch - for example, I just replaced the call in AlertLogPkg to initialize to LocalInitialize and the method initialize now calls LocalInitialize.  That should close out any potential language issues.

cmarqu, 2016-02-10:
Thanks very much, that sounds good.

JimLewis, 2016-02-20:
I just created a Dev_Cadence branch with this patch for Cadence and a number of others.  Give it a try and let me know the next set of issues.  I would like the Dev_Cadence branch to have all of the patches necessary to get the library working under Cadence simulators.

closed 2016-02-20 by JimLewis