OSVVM/OSVVM #7: Support for Cadence Incisive

cmarqu, 2016-02-22:
Alright, attached is the list of errors I get with the Makefile from https://github.com/JimLewis/OSVVM/pull/6. This is with irun 15.10-s013: [incisive.txt](https://github.com/JimLewis/OSVVM/files/140354/incisive.txt)

JimLewis, 2016-02-22:
Hi Coliin
I tried to address all of these in the latest push to the Dev_Cadence branch.
Send me the next batch when you get a chance.  I am hoping to get a 
compile run from another colleague in the US so maybe I will have a
second update.  
Jim

cmarqu, 2016-02-23:
Thanks, compilation works without problems in IUS version 15.10 now!
With IUS 14.20-s014 I get:

```
[...]
OsvvmContext.vhd:
context OsvvmContext is
|
ncvhdl_p: *E,EXPACE (OsvvmContext.vhd,47|0): expecting a library unit [11.1].
    use OSVVM.MemoryPkg.all ;
                              |
ncvhdl_p: *E,EXPACE (OsvvmContext.vhd,56|30): expecting a library unit [11.1].

ncvhdl_p: *E,UXPEOF (OsvvmContext.vhd,59): unexpected end of file.
        errors: 3, warnings: 0
irun: *E,VHLERR: Error during parsing VHDL file (status 1), exiting.
```

cmarqu, 2016-02-23:
Sorry, I spoke too soon - I get the same errors about ``context`` in both simulator versions.

cmarqu, 2016-02-23:
I created a pull request for Makefile improvements for simulating the demos. This needs further changes in order to have them compile: 
- ``std.env.stop`` needs to be written as ``std.env.stop(0) ;`` in ``AlertLog_Demo_*.vhd``
- Usage of library SynthWorks should be replaced by library ``osvvm`` in ``Demo_Rand.vhd``

In simulation of ``AlertLog_Demo_Global``, this then leads to:
```
ncsim> run
ncsim: *E,TRDANGD: dangling (deallocated) pointer dereference.
          File: ./AlertLogPkg.vhd, line = 1394, pos = 19
          Time: 0 FS + 0

ncsim: *E,PNOOBJ: Path element could not be found: :SetLogEnable(LogType,BOOLEAN).
./AlertLogPkg.vhd:1394         AlertLogPtr(i).LogEnabled(Level) := Enable ;
ncsim> exit
```

JimLewis, 2016-02-23:
Updated to Try4 (please skip Try3).  It looks like the allocations in AlertLogPkg have not happened yet.  Not good.  Added debug code to AlertLogPkg.  Added wait for 1 ns to Demo programs to give it time to happen.  If it does not happen, it never will at that point.   If the call to SetLogEnable returns 

Did you by chance try AlertLogPkg_alt1?  It tries to fix LocalInitialize by replicating the code.  It may be worth trying if the above does not work.  However, it may also bring us back to compile errors.  

If neither of the above work, I have a couple of other things to try.  
One alternative I could try is to create a regular function that returns a pointer to the initialized AlertLogPtr. 

The fall back alternative is not to initialize the data structure and require the user to do it with a call to InitializeAlertLogStruct.

cmarqu, 2016-02-23:
AlertLogPkg_alt1.vhd gives me
```
AlertLogPkg_alt1.vhd:
    constant CONSTRUCT_ALERT_DATA_STRUCTURE : boolean := LocalInitialize ; 
                                                                       |
ncvhdl_p: *E,IPMCAL (AlertLogPkg_alt1.vhd,1106|71): Illegal protected method call or member access as per LRM [10.3].
```
Will try the rest now.

cmarqu, 2016-02-23:
Using AlertLogPkg.vhd, it's pretty much the same:
```
ncsim> run
ncsim: *E,TRDANGD: dangling (deallocated) pointer dereference.
          File: ./AlertLogPkg.vhd, line = 1395, pos = 22
          Time: 0 FS + 0

ncsim: *E,PNOOBJ: Path element could not be found: :SetLogEnable(LogType,BOOLEAN).
./AlertLogPkg.vhd:1395         if AlertLogPtr(i) /= NULL then
ncsim> exit
```

(Had to make a small patch:
```
@@ -1396,6 +1396,7 @@ package body AlertLogPkg is
           AlertLogPtr(i).LogEnabled(Level) := Enable ;
         else
           Alert("SetLogEnable: LogID " & integer'image(i) & " not initialized ") ;
+        end if ; 
       end loop ; 
     end procedure SetLogEnable ;    
 ```
)

JimLewis, 2016-02-23:
Sorry about not compiling Try3 first.  I rectified that in Try4.

JimLewis, 2016-02-23:
I think I found it.  I just pushed some changes.  In the demo programs there was a concurrent call to SetLogEnable.  I put it into a process.  I also put a wait for 1 ns before it.  SetLogEnable makes a call into the AlertLog Structure.  If Cadence tries to optimize this and evaluate it during initialization, then there would be a race condition between the data structure being constructed in the package body and the call.  Other simulators had issues with concurrent method calls in the past.  

If you would, please try this first.

cmarqu, 2016-02-23:
Hmm, for some reason, a simple git pull didn't give me try4 apparently. I don't really know git...

Ok, so I used AlertLogPkg_try6.vhd and uncommented the line ``InitializeAlertLogStruct ;``. This gave me:
```
ncsim> run
ncsim: *F,INTERR: INTERNAL EXCEPTION
Observed simulation time : 0 FS + 0
-----------------------------------------------------------------
The tool has encountered an unexpected condition and must exit.
Contact Cadence Design Systems customer support about this
problem and provide enough information to help us reproduce it,
including the logfile that contains this error message.
  TOOL: ncsim   15.10-s013
  HOSTNAME: l38
  OPERATING SYSTEM: Linux 2.6.18-406.el5 #1 SMP Tue Jun 2 17:25:57 EDT 2015 x86_64
  MESSAGE: T(0): sv_seghandler - trapno -1 addr(0x10)
        Subprogram body: NEWALERTLOGREC (line: 1003 in design unit OSVVM.ALERTLOGPKG:body)
-----------------------------------------------------------------
csi-ncsim - CSI: Cadence Support Investigation, sending details to /home/marquardt/tools/OSVVM/ncsim_29727.err
csi-ncsim - CSI: investigation complete, send /home/marquardt/tools/OSVVM/ncsim_29727.err to Cadence Support
make: *** [AlertLog_Demo_Global] Error 255
```

JimLewis, 2016-02-23:
I am leaving on a month long trip on Friday.  To accelerate the debug process, I included a couple of variations on AlertLogPkg:

Make sure to be using the updated  Demo programs as the concurrent call to SetLogEnable had to be the one generating the errors, as otherwise, the error would have pointed to SetAlertLogName.  If the failure switches to be on SetAlertLogName, add a wait for 1 ns before it.

In AlertLogPkg_try5.vhd, I changed the allocation to be in a function call rather than directly initialize the object as in the current AlertLogPkg.vhd.  If this works in Cadence, this is a template that I would like to use to refactor some of the code, and hopefully, work the Cadence and Standard variations of this code to match in this part of the code.   If the current AlertLogPkg.vhd works, this one is worth pursuing.  If not, it does the allocation different, so it may work where the current one does not, I am not sure - this is kind of like debugging an FPGA in the lab where you only have a high level product spec.  

AlertLogPkg_try6.vhd is a last resort solution.  Don't bother with this step if either AlertLogPkg.vhd or AlertLogPkg_try5.vhd work.  In this version, I removed the allocation from the AlertLogPkg.  There is a call to InitializeAlertLogStruct that is commented out in AlertLog_Demo_Global.vhd that will need to be uncommented to test this one.  AlertLog_Demo_Hierarchy.vhd will not work with try6.  There are calls to GetAlertLogID in declarative regions (assigned to constants) that will not work since the structure is not allocated until run time.  If this ends up the path we have to take to get Cadence working, I will update that demo program.

cmarqu, 2016-02-23:
For completeness' sake:
```
AlertLogPkg_try5.vhd:
    variable AlertLogPtr  : AlertLogArrayPtrType := InitializeAlertLogPtr ;
                                                                        |
ncvhdl_p: *E,IPMCAL (AlertLogPkg_try5.vhd,599|72): Illegal protected method call or member access as per LRM [10.3].
[...]
```

cmarqu, 2016-02-23:
With AlertLogPkg.vhd:
```
ncsim> run
ncsim: *F,INTERR: INTERNAL EXCEPTION
Observed simulation time : 0 FS + 0
-----------------------------------------------------------------
The tool has encountered an unexpected condition and must exit.
Contact Cadence Design Systems customer support about this
problem and provide enough information to help us reproduce it,
including the logfile that contains this error message.
  TOOL: ncsim   15.10-s013
  HOSTNAME: l38
  OPERATING SYSTEM: Linux 2.6.18-406.el5 #1 SMP Tue Jun 2 17:25:57 EDT 2015 x86_64
  MESSAGE: T(0): sv_seghandler - trapno -1 addr(0x40)
        Subprogram body: GET (line: 85 in design unit OSVVM.NAMEPKG:body)
-----------------------------------------------------------------
csi-ncsim - CSI: Cadence Support Investigation, sending details to /home/marquardt/tools/OSVVM/ncsim_30099.err
csi-ncsim - CSI: investigation complete, send /home/marquardt/tools/OSVVM/ncsim_30099.err to Cadence Support
make: *** [AlertLog_Demo_Global] Error 255
```

JimLewis, 2016-02-23:
The issues with AlertLogPkg_try5.vhd is a show stopper for that path.

cmarqu, 2016-02-23:
I can try playing with your info and the contents of the *.err files and see if I get anywhere. Have a good trip!

JimLewis, 2016-02-23:
I haven't given up yet.  I think I understand its current issue with AlertLogPkg.  I will have a patch shortly.

JimLewis, 2016-02-23:
Just pushed a change that should address the issues that reported against NamePkg.  

I suspect there will be more issues when it gets to ReportAlerts.  I am working on that change now.

cmarqu, 2016-02-23:
Ok, I am using AlertLogPkg.vhd now and am not calling ``InitializeAlertLogStruct``.
This gives me 
```ncsim: *E,TRDANGD: dangling (deallocated) pointer dereference.
          File: ./AlertLogPkg.vhd, line = 1018, pos = 28
          Time: 0 FS + 0

ncsim: *E,PNOOBJ: Path element could not be found: :SetAlertLogName.
./AlertLogPkg.vhd:1018       Deallocate(AlertLogPtr(ALERTLOG_BASE_ID).Name) ; 
```
Adding ``wait for 1 ns ;`` before ``SetAlertLogName("AlertLog_Demo_Global") ; `` gives me
```
ncsim: *E,TRDANGD: dangling (deallocated) pointer dereference.
          File: ./AlertLogPkg.vhd, line = 1400, pos = 22
          Time: 1 NS + 0

ncsim: *E,PNOOBJ: Path element could not be found: :SetLogEnable(LogType,BOOLEAN).
./AlertLogPkg.vhd:1400         if AlertLogPtr(i) /= NULL then
```

cmarqu, 2016-02-23:
I have to leave now and won't be in tomorrow. Maybe someone else can pick this up in the meantime.

JimLewis, 2016-02-23:
Thanks.  It is getting closer.  I have to meditate on the implications of those last issues.

JimLewis, 2016-02-23:
The latest error looks like what would be generated by the AlertLogPkg_try6 without the initialization.    

I am hoping to get back to the main AlertLogPkg.vhd version that generated the following.  I think it is the one with any "_" in it.   

> ncsim> run
ncsim: *F,INTERR: INTERNAL EXCEPTION
Observed simulation time : 0 FS + 0
-----------------------------------------------------------------
The tool has encountered an unexpected condition and must exit.
Contact Cadence Design Systems customer support about this
problem and provide enough information to help us reproduce it,
including the logfile that contains this error message.
  TOOL: ncsim   15.10-s013
  HOSTNAME: l38
  OPERATING SYSTEM: Linux 2.6.18-406.el5 #1 SMP Tue Jun 2 17:25:57 EDT 2015 x86_64
  MESSAGE: T(0): sv_seghandler - trapno -1 addr(0x40)
        Subprogram body: GET (line: 85 in design unit OSVVM.NAMEPKG:body)
-----------------------------------------------------------------
csi-ncsim - CSI: Cadence Support Investigation, sending details to /home/marquardt/tools/OSVVM/ncsim_30099.err
csi-ncsim - CSI: investigation complete, send /home/marquardt/tools/OSVVM/ncsim_30099.err to Cadence Support
make: *** [AlertLog_Demo_Global] Error 255

I think I have fixed it in the most current AlertLogPkg.vhd.  

I think your compiles have shown that all of the "_" paths have issues with them.  Although I am concerned that try6 also shows that I will need to look at the deallocate and then initialize process in Cadence at some point in time - however that is not that important just now, and maybe we can leave for them to fix.

cmarqu, 2016-02-25:
Ok, current status is
```
ncsim: *E,TRDANGD: dangling (deallocated) pointer dereference.
          File: ./AlertLogPkg.vhd, line = 1028, pos = 28
          Time: 1 NS + 0

ncsim: *E,PNOOBJ: Path element could not be found: :SetAlertLogName.
./AlertLogPkg.vhd:1028       Deallocate(AlertLogPtr(ALERTLOG_BASE_ID).Name) ; 
```

JimLewis, 2016-02-25:
Hi Colin,
I just pushed Try 9, but I just thought of something else I should check in the debug code.  Give me another minute and I will update it.
Jim

JimLewis, 2016-02-25:
Hi Colin,
I pushed Try 10.  

I note that one time it looked like it failed in a call to Alert.  That is much further than this branch.  It could have happened with a wait for 1 ns before the SetAlertLogName, like I did here.  So maybe this will address it. 

Jim

cmarqu, 2016-02-25:
Hi Jim,

try10 gives me
```
ncsim: *E,TRDANGD: dangling (deallocated) pointer dereference.
          File: ./AlertLogPkg.vhd, line = 1030, pos = 20
          Time: 1 NS + 0

ncsim: *E,PNOOBJ: Path element could not be found: :SetAlertLogName.
./AlertLogPkg.vhd:1030       if AlertLogPtr(ALERTLOG_BASE_ID) /= NULL then
```

JimLewis, 2016-02-25:
Hi Colin, 
Based on that, they are not building the data structure.  In Try 11, I added more debugging code, but it is not going to tell us anything we do not currently know.  I added it to help Cadence debug if we can engage them.

If you want to go further, I would strip the initialization back out.  This involves changing lines 537 and 538 to: 
  variable NumAlertLogIDsVar  : AlertLogIDType := 0;  
  variable NumAllocatedAlertLogIDsVar : AlertLogIDType := 0 ;

And 545 to: 
  variable AlertLogPtr  : AlertLogArrayPtrType ;

And then on line 94 of AlertLog_Demo_Global.vhd uncommenting the call to InitializeAlertLogStruct.  

It is possible that this could fix it, however, reviewing the previous information, it looks like it was failing at the point at which items were being attempted to be added to the data structure.

Since we have it to the point of run time failures, perhaps we can convince the Cadence folks to take a look at it.  I will give it a shot on my side.

Best Regards,
Jim

cmarqu, 2016-02-25:
Unmodified try11 indeed looks like before.
I now did the changes, and it's back to an internal exception (which I expect should never happen, regardless of whatever "strange" code it might encounter):
```
ncsim: *F,INTERR: INTERNAL EXCEPTION
Observed simulation time : 0 FS + 0
-----------------------------------------------------------------
The tool has encountered an unexpected condition and must exit.
Contact Cadence Design Systems customer support about this
problem and provide enough information to help us reproduce it,
including the logfile that contains this error message.
  TOOL: ncsim   15.10-s013
  HOSTNAME: l38
  OPERATING SYSTEM: Linux 2.6.18-406.el5 #1 SMP Tue Jun 2 17:25:57 EDT 2015 x86_64
  MESSAGE: T(0): sv_seghandler - trapno -1 addr(0x10)
        Subprogram body: NEWALERTLOGREC (line: 1060 in design unit OSVVM.ALERTLOGPKG:body)
-----------------------------------------------------------------
csi-ncsim - CSI: Cadence Support Investigation, sending details to /home/marquardt/tools/OSVVM/ncsim_2107.err
csi-ncsim - CSI: investigation complete, send /home/marquardt/tools/OSVVM/ncsim_2107.err to Cadence Support
```

JimLewis, 2016-02-25:
NewAlertLogRec is in the call path of InitializeAlertLogStruct.  First the outer data structure is alloacated then there are three calls to NewAlertLogRec which tries to add the basic elements to it. 

Unfortunately, this failure mode is the same as the Try 10, as soon as it tries to access the data structure, it dies.  

Please reach out to Cadence support for help.  I am reaching out to some of the contacts I have.

JimLewis, 2016-02-25:
Have you tried Demo_Rand.vhd?  It does not have any pointers, so it should be fine.

JimLewis, 2016-02-25:
Have you added references to the files that is not in _github?  
Specifically ieee_proposed.standard_additions_c.vhdl?

cmarqu, 2016-02-25:
Ok, I will contact Cadence.
Demo_Rand runs nicely indeed!

cmarqu, 2016-02-25:
No, I'm purely using the GitHub clone.

JimLewis, 2016-05-31:
Anyone one test with a more recent Cadence release?

cmarqu, 2016-06-01:
With IUS 15.20-s003 and using none of the try* versions, I still get:
```
ncsim: *E,TRDANGD: dangling (deallocated) pointer dereference.
          File: ./AlertLogPkg.vhd, line = 1032, pos = 22
```

JimLewis, 2018-07-04:
Cadence last indicated that you need to move to their new simulator to get support for protected types.

closed 2018-07-04 by JimLewis

cmarqu, 2018-07-04:
Alright, I'll try with Xcelium soon and open a new ticket then.