OSVVM/OSVVM #9: Support Names for Bins

tmeissner, 2016-03-02:
It would be nice, if we could set names for Bins.
Maybe we have a coverage model for a simple 3 state FSM:

```vhdl
type t_fsm_state is (IDLE, ADDR, DATA);
signal s_fsm_state, s_fsm_state_prev : t_fsm_state;

shared variable sv_fsm_cov : CovPType;

sv_fsm_cov.AddCross(GenBin(t_fsm_state'pos(IDLE)), GenBin(t_fsm_state'pos(ADDR));
sv_fsm_cov.AddCross(GenBin(t_fsm_state'pos(ADDR)), GenBin(t_fsm_state'pos(DATA));
sv_fsm_cov.AddCross(GenBin(t_fsm_state'pos(ADDR)), GenBin(t_fsm_state'pos(IDLE));
sv_fsm_cov.AddCross(GenBin(t_fsm_state'pos(DATA)), GenBin(t_fsm_state'pos(IDLE));
sv_fsm_cov.AddCross(ALL_ILLEGAL, ALL_ILLEGAL);

...

sv_fsm_cov.ICover(t_fsm_state'pos(s_fsm_state_prev), t_fsm_state'pos(s_fsm_state));

...

sv_fsm_cov.WriteBin;

```

At the moment, the integer values are shown in the report like this:

```
# %%WriteBin: 
# %% Bin:(0) (1)   Count = 1  AtLeast = 1
# %% Bin:(1) (0)   Count = 1  AtLeast = 1
# %% Bin:(1) (2)   Count = 1  AtLeast = 1
# %% Bin:(2) (0)   Count = 1  AtLeast = 1
```

It would be nicer if we had names for Bins.
Maybe, the report could use them like this:

```
# %%WriteBin: 
# %% Bin:(IDLE) (ADDR)   Count = 1  AtLeast = 1
# %% Bin:(ADDR) (DATA)   Count = 1  AtLeast = 1
# %% Bin:(ADDR) (IDLE)   Count = 1  AtLeast = 1
# %% Bin:(DATA) (IDLE)   Count = 1  AtLeast = 1
```

That could be realised with overloading maybe.
If you give a name with the Bin creation, we have such a new report,
if not, we have the old above with integers only.

Furthermore a name field of type string has to be added to the CovBinBaseType
and the CovMatrix* record types.

tmeissner, 2016-03-02:
Ah, in the current version, you can call AddCross() with a Name parameter. So, that would help in the example above. I didn't know that feature because my test was using an older OSVVM version which came with Modelsim 10.4 :(

Paebbels, 2016-03-02:
Yes, OSVVM got named Bin support in version 2014.07 :)

```VHDL
procedure AddBins(Name : String; AtLeast : integer; Weight : integer ; CovBin : CovBinType);
-- plus many overloads ...
```

tmeissner, 2016-03-02:
Yep, for FSM coverage it can look like these:

```vhdl
sv_fsm_cov.AddCross("IDLE->ADDR",
  GenBin(t_fsm_state'pos(IDLE)),
  GenBin(t_fsm_state'pos(ADDR));
sv_fsm_cov.AddCross("ADDR->DATA",
  GenBin(t_fsm_state'pos(ADDR)),
  GenBin(t_fsm_state'pos(DATA));
sv_fsm_cov.AddCross("ADDR->IDLE",
  GenBin(t_fsm_state'pos(ADDR)),
  GenBin(t_fsm_state'pos(IDLE));
sv_fsm_cov.AddCross("DATA->IDLE",
  GenBin(t_fsm_state'pos(DATA)),
  GenBin(t_fsm_state'pos(IDLE));
sv_fsm_cov.AddCross(ALL_ILLEGAL, ALL_ILLEGAL);
```

generates following report:

```
# %%WriteBin: 
# %% IDLE->ADDR  Bin:(0) (1)   Count = 1  AtLeast = 1
# %% ADDR->DATA  Bin:(1) (0)   Count = 1  AtLeast = 1
# %% ADDR->IDLE  Bin:(1) (2)   Count = 1  AtLeast = 1
# %% DATA->IDLE  Bin:(2) (0)   Count = 1  AtLeast = 1
```

Great :)

closed 2016-05-31 by JimLewis