OSVVM/OSVVM #10: Assignment Of Access Values

peteut, 2016-04-04:
In `AlertLogPkg.vhd` a some assignments of access values w/o `.all.`. This causes a error using `nvc`.

Example: 

```vhdl
entity access_assignment is
end entity;
 
architecture test of access_assignment is
type rec_t is record
  foo: integer;
end record rec_t;
type rec_ptr_t is access rec_t;
type rec_arr_t is array (integer range <>) of rec_ptr_t;type rec_arr_ptr_t is access rec_arr_t;
procedure ok_procedure is
variable rec_ptr: rec_arr_ptr_t;
begin
  rec_ptr(0) := new rec_t'(foo => 42);
  rec_ptr(0).all.foo := 2;
end;
procedure nok_procedure is
variable rec_ptr: rec_arr_ptr_t;
begin
  rec_ptr(0) := new rec_t'(foo => 42);
  rec_ptr(0).foo := 2; -- <- error here
end;
begin
end architecture;
```

`nvc` error:
```
** Error: expected record type but found REC_PTR_T
        File access_bug.vhd, Line 15
        rec_ptr(0).foo := 2; -- <- error here
        ^^^^^^^^^^
```

According to my literature, *The Designer's Guide to VHDL - Third Edition* the `.all.` is required and hence not a bug of `nvc`[1].

[1] The Designer's Guide to VHDL Third Edition, section 15.1.2 Assignment and Equality of Access Values

tmeissner, 2016-04-05:
Mhm, one chapter later, they describe

> We have seen that we can access a record object pointed to by bus_stimulus using the notation “bus_stimulus.all”. If we want to refer to the stimulus_time element, we could write “bus_stimulus.all.stimulus_time”.
> 
> In practice, we usually want to refer either to the pointer itself or to an element of the record, and rarely to the record as a whole. For this reason, VHDL allows us to write “bus_stimulus.stimulus_time” to refer to the record element. Whenever we select a record element name after an access variable name, we automatically follow the pointer to get to the record.

Isn't that the case with the code above?

peteut, 2016-04-05:
Indeed, `.all:` is not required to access elements. The compiler does not lower it correctly. Thanks for clarification!

closed 2016-04-05 by peteut