OSVVM/OSVVM #19: Can OSVVM provide a "vhdl_analyze_order" file?

Paebbels, 2016-12-07:
Xilinx and some other vendors provide a file called `vhdl_analyze_order` per VHDL library, containing one source file per line. This file can be read by Tcl, Bash or what ever language to execute the compile commands for a VHDL library in the right order. Alternative names might be: `analyze_order` or `compile_order`.

This solution has several advantages:
- it is tool independent (e.g. Tcl compile scripts must be provided per simulator)
- all files are stored with relative path, so scripts can add the appropriate absolute path if needed
- if OSVVM adds more files or changes the compile order, only one file needs to be adjusted, which is also under version control and thus linked to the current release
- pre-compile scripts (e.g. shipped with GHDL or PoC) don't need to implement version specific switches to chose the right file names and right compile order

**Example:**
```
NamePkg.vhd
OsvvmGlobalPkg.vhd
TextUtilPkg.vhd
TranscriptPkg.vhd
AlertLogPkg.vhd
MemoryPkg.vhd
MessagePkg.vhd
SortListGenericPkg.vhd
SortListPkg.vhd
SortListPkg_int.vhd
RandomBasePkg.vhd
RandomPkg.vhd
CoveragePkg.vhd
ScoreboardGenericPkg.vhd
ScoreboardPkg.vhd
OsvvmContext.vhd
```

**Example Bash script to read such a file:**
```Bash
Library="xilinxcorelib"
	
# append absolute source path
SourceFiles=()
while IFS= read -r File; do
  SourceFiles+=("$SourceDirectory/XilinxCoreLib/$File")
done < <(grep --no-filename -R '^[a-zA-Z]' "$SourceDirectory/XilinxCoreLib/vhdl_analyze_order")

GHDLCompilePackages
```
Source: https://github.com/tgingold/ghdl/blob/master/libraries/vendors/compile-xilinx-ise.sh?ts=2#L366-L376

JimLewis, 2016-12-07:
I included a tcl based do file with the current version, but I like the idea of a compile order file, perhaps osvvm_compile_order.   

Do you have a similar file that you use for running tests?   I have a run_all.sh script, but it uses specific names in it.   It is also time to look at VUNIT.

Paebbels, 2016-12-07:
We could also move this discussion to Gitter :)

I don't think a prefix like "osvvm_" is needed, because users hopefully copy/install OSVVM into a separate subdirectory already called "osvvm". That's atleast the case for PoC and VUnit, which embed OSVVM as Git submodules.

With the Bash snippet presented above, you could integrate the compile order into your `run_all.sh` script.

-----------------
PoC itself uses far more advanced compile order files ;)
We have `*.files` files for IP core and also for external library. These files support:
- conditional compilation based on:
  - tool chains,
  - VHDL version,
  - environment (sim vs. synth).
- We can also reference existing pre-compile libraries (useful with GHDL or vsim) and we can
- include other *.files files.

This is the `osvvm.files` file:
```
path PreCompiled =      ${CONFIG.DirectoryNames:PrecompiledFiles}
path OSVVM_Directory =  "osvvm"

if (VHDLVersion < 2008) then
  report "OSVVM requires VHDL-2008; See OSVVM documentation for VHDL-2002 support."
elseif (VHDLVersion = 2008) then
  if (Tool = "GHDL") then
    path GHDLPath = (PreCompiled / ${CONFIG.DirectoryNames:GHDLFiles})
    if ?{((GHDLPath / OSVVM_Directory) / "v08/osvvm-obj08.cf")} then
      library osvvm	GHDLPath
    else
      report "No precompiled OSVVM library for GHDL found."
    end if
  elseif (Tool = "Mentor_vSim") then
    path OSVVM_LibraryPath = (PreCompiled / (${CONFIG.DirectoryNames:QuestaSimFiles} / OSVVM_Directory))
    if ?{OSVVM_LibraryPath} then
      library osvvm	OSVVM_LibraryPath
    else
      report "No precompiled OSVVM library for QuestaSim/ModelSim found."
    end if
  else
    # TODO self-compile section?
    vhdl	osvvm	"lib/osvvm/NamePkg.vhd"
    vhdl	osvvm	"lib/osvvm/OsvvmGlobalPkg.vhd"
    vhdl	osvvm	"lib/osvvm/TextUtilPkg.vhd"
    vhdl	osvvm	"lib/osvvm/TranscriptPkg.vhd"
    vhdl	osvvm	"lib/osvvm/AlertLogPkg.vhd"
    vhdl	osvvm	"lib/osvvm/MemoryPkg.vhd"
    vhdl	osvvm	"lib/osvvm/MessagePkg.vhd"
    vhdl	osvvm	"lib/osvvm/SortListPkg_int.vhd"
    vhdl	osvvm	"lib/osvvm/RandomBasePkg.vhd"
    vhdl	osvvm	"lib/osvvm/RandomPkg.vhd"
    vhdl	osvvm	"lib/osvvm/CoveragePkg.vhd"
    vhdl	osvvm	"lib/osvvm/OsvvmContext.vhd"
  end if
else
  report "VHDL version not supported by OSVVM."
end if
```
Source: https://github.com/VLSI-EDA/PoC/blob/master/lib/OSVVM.files?ts=2

JimLewis, 2016-12-07:
I like this better than gitter since all of the thoughts on this are recorded under this topic.

JimLewis, 2016-12-08:
I like either a file list or a simple compile like the osvvm.do that was released in with OSVVM.   

If I were to start to add checks, such as you showed, it will keep reasonable things from happening.  For example, someone wants to update a vendor version to the newest OSVVM release.

Paebbels, 2016-12-08:
My problem with *.do files is that these are hard to parse, so I can't reuse these files for Bash or Python based compilations. So I prefer a plain file list (one file per line).

Paebbels, 2018-05-06:
Any news on that issue?

JimLewis, 2018-05-06:
Hi Patrick,
This is on the someday, maybe list.   It is my intent to only provides items like this that are verified to be correct.   As you know my focus right now is on the AXI4 family of models.  

You state, "My problem with *.do/*.tcl files is that these are hard to parse".   Let me counter that with a simple algorithm:
1.  Delete all comment lines
2.  Change all spaces to new line
3.  Change all "/" to new line
4.  Keep only lines that have "vhd" in them.

Now lets implement that algorithm in bash/linux:
sed -e "s/#.*$//" osvvm.do | sed -e "s/ /\n/g" | sed -e "s/\//\n/" | grep vhd

With some small tweaks, this can be made to turn your PoC script/descriptor files into compile order files.

I suspect some day soon I may require something different so I can include other tools.   OTOH, I am also getting better at *.tcl so who knows.   

Best Regards,
Jim

Paebbels, 2018-05-06:
Sorry, but this not an acceptable solution, because I'm highly dependent on you Tcl syntax.

Moreover, your suggested solution works only on a subset of platforms that PoC annd GHDL support. E.g. Bash on Linux, Bash in MinGW. This will not work in native Windows with e.g. PowerShell.

**Reading a compile order file in Tcl can be done so:**

    quietly set fp [open "$source_path/src/compile_order.txt" r]
    quietly set file_data [read $fp]
    quietly set lib_name [lindex $file_data 2]
    close $fp

Source: https://github.com/UVVM/UVVM/blob/master/bitvis_vip_sbi/script/compile_src.do#L91-L97

**Looping the lines and calling vcom can be done so:**

    echo "\n\n\n=== Compiling $lib_name source\n"
    quietly set idx 0
    foreach item $file_data {
      if {$idx > 2} {
        echo "eval vcom  $compdirectives  $source_path/src/$item"
        eval vcom  $compdirectives  $source_path/src/$item
      }
      incr idx 1
    }

Source: https://github.com/UVVM/UVVM/blob/master/bitvis_vip_sbi/script/compile_src.do#L133-L141

-------------------
Shall I create a pull request for this change? I can test the scripts here on ModelSim, Riviera-PRO and Active-HDL.

JimLewis, 2018-05-06:
I have something like that for running the regression suite.   However, the compile script for the library should be simple and readable, so that basic users can understand it.   

The simple script for OSVVM will be preserved.  Hence, any compile.txt will be an addition and need to be independently validated.   

The bigger issue with compile.txt and OSVVM is that OSVVM has the VendorCovApiPkg which is intended to have vendor specific implementations of subprograms that log functional coverage into the tool.   The intent is that there is a file for each vendor and/or simulator.   In the simple script, there are comments to explain this.   

What do you do with a file that keeps it being a simple file and not a program/script?  In your PoC "file" it looks more like you swapped TCL for something else (Python?).

Paebbels, 2018-05-06:
For vendor specific files like used by OSVVM coverage package, the order does not change. Thus, one could use a simple template description:

    file1.vhdl
    file2$vendor.vhdl

A line-by-line reading script could exchange the `$vendor` variable with `Aldec`, `Mentor`, ... or just an empty string to fallback to a default package. Any scripting language can handle such variable based replacements.

E.g. UVVM uses the first comment line to specify the VIP's VHDL library name to compile into.

For understandability by basic users, I think Tcl is the wrong approach :). But if a basic Tcl user can read a for-each loop, it is no difference if the loop source is an array or a file. A good descriptive comment may help to understand such a loop. If a user doesn't know for-each loops, he will have bigger problems in understanding scripting at all.

-----------
The `*.files` syntax is a domain specify language with conditionals, includes, ... It is not intended to be read by basic users. The language needs an extensive parser written in Python, because it handles complex variable replacements, expression evaluation, function calls, file checks, etc.

Long time ago, it was derived from the simple iSim / xSim `*.prj` files.

    <language> <library> <path/to/file>
    <language> <library> <path/to/file>

Example:

    vhdl   poc     src/arith/arith_prng.vhdl

Nowadays, I can describe complex dependencies, different file sets based on vendor tools, VHDL version, workarounds, etc. But this capability is not needed by OSVVM, right?

JimLewis, 2020-02-06:
Hi Patrick,
I am looking to close issues.   Have you looked at OSVVM's *.pro files.  
They look like:
library osvvm_TbAxi4Lite
analyze TestCtrl_e.vhd
analyze TbAxi4Lite_map_records.vhd
analyze TbAxi4Lite_BasicReadWrite.vhd
simulate TbAxi4Lite_BasicReadWrite

they also support "include" and "build"
include ../common/common.pro
include ./Axi4Lite.pro
include ./testbench/testbench.pro

Paebbels, 2020-02-08:
@JimLewis this is a nice an clean structure.

* This this format somewhere documented?  
  If not, please give an example for `build`.
* Does `simulate` support, entities and configurations?  
  How to pass the library name to `simulate`?

JimLewis, 2020-05-09:
Documentation in VerificationIP/Scripts/ScriptApproach.pdf

closed 2020-05-09 by JimLewis