OSVVM/OSVVM #20: Scoreboard for VHDL 2002

LarsAsplund, 2016-12-19:
[ScoreboardGenericPkg.vhd](https://github.com/OSVVM/OSVVM/blob/master/ScoreboardGenericPkg.vhd) contains comments on how to modify the file for simulators not supporting package generics (latest binary/stable release of GHDL for example). I would prefer if such a modification is provided as a separate implementation in the same way that `VendorCovApiPkg` has several implementations. This would remove need for users to create and maintain their own implementations.

JimLewis, 2016-12-19:
The distribution currently provides instances for ScoreboardPkg_int.vhd and ScoreboardPkg_slv.vhd.   Are those the ones you are using or anticipate using?   Hence, do you need ScoreboardPkg_int_c.vhd and ScoreboardPkg_slv_c.vhd?

On my projects, I typically use a record type that is specific to the interface I am working with - for example, my UART receiver will have a data value and the expected status (parity error, stop error, and break error).   Hence, for my projects, I have to create a custom package instance for the interface, ScoreboardPkg_UartRx.vhd.    

In reality, there is little difference in effort between creating a package instance ScoreboardPkg_UartRx.vhd and creating a custom package with subtypes and aliases named ScoreboardPkg_UartRx_c.vhd.    Where the real advantage of generic packages comes into play is when the package changes.   

Going further I think that GHDL is currently the only simulator that I know of that supports protected types but does not support generic types on packages.   And the reports I am hearing about GHDL from Patrick are quite positive.    So anything I provide would only be useful for a short amount of time. 
In reality, there is little difference in effort between creating a package instance ScoreboardPkg_UartRx.vhd and creating a custom package with subtypes and aliases named ScoreboardPkg_UartRx_c.vhd.    Where the real advantage of generic packages comes into play is when the package changes.   

Going further I think that GHDL is currently the only simulator that I know of that supports protected types but does not support generic types on packages.   And the reports I am hearing about GHDL from Patrick are quite positive.    So anything I provide would only be useful for a short amount of time. 
In reality, there is little difference in effort between creating a package instance ScoreboardPkg_UartRx.vhd and creating a custom package with subtypes and aliases named ScoreboardPkg_UartRx_c.vhd.    Where the real advantage of generic packages comes into play is when the package changes.   

Going further I think that GHDL is currently the only simulator that I know of that supports protected types but does not support generic types on packages.   And the reports I am hearing about GHDL from Patrick are quite positive.    So anything I provide would only be useful for a short amount of time.

LarsAsplund, 2016-12-21:
My primary concern is the build support for OSVVM in VUnit. OSVVM became a pure submodule to VUnit without any modifications earlier this year. Since then all that VUnit provides is the ability to do `add_osvvm()` in the user Python script such that OSVVM is compiled into the project without the user having to worry about how OSVVM is compiled to match the used simulator. Now that some simulator versions do not support package generics (VUnit supports more than the latest version) I see two options
1. VUnit do not compile the scoreboard files when users of these simulators use the `add_osvvm()` method. If they want to use the scoreboard they will have to modify OSVVM according to your suggestions and then add the OSVVM files to the VUnit project manually. Pros: The scoreboard will work as you intended. Cons: Less user friendly and less convenient when upgrading OSVVM version
2. OSVVM provides a number of standard type scoreboards, at least one for `std_logic_vector`. VUnit would compile these when calling `add_osvvm()` with one of the concerned simulators. Users create their own scoreboards by converting their custom type into one of the standard type scoreboards. For example by concatinating the bits of your UART example type. Pros: There is a clear separation between user code and third party libraries and there are no issues when OSVVM is updated in a backward compatible manner. Cons: The outputs of the `to_string` functions will not be as good looking as they can be.

I'm ok with both approaches. Currently I use the first method for GHDL and older versions of Aldec's simulators but I would update to the second if you provide VHDL-2002 scoreboards

When it comes to GHDL I'm not sure when there will be an official version supporting package generics. The latest, 0.33, is over a year old. I think many people avoid 0.34dev as long as no binary releases are provided.

JimLewis, 2019-02-05:
Providing copies of VHDL-2002 scoreboards is obtainable by the directions.   For now we will leave it at that and close this issue.

closed 2019-02-05 by JimLewis