OSVVM/OSVVM #22: Use of overloaded & in CoveragePkg

MichaelK92, 2017-06-27:
Hi!

When compiling the latest osvvm using the ghdl/lib/ghdl/vendors/compile-osvvm.sh script I got the following error:

```
Analyzing package '/opt/osvvm/CoveragePkg.vhd'
/opt/osvvm/CoveragePkg.vhd:1477:33: operator ""&"" is overloaded
/opt/osvvm/CoveragePkg.vhd:1477:33: possible interpretations are:
*std_standard*:1:1: function "&" [string, character return string]
*std_standard*:1:1: function "&" [string, string return string]
```

The culprit is here:
<pre>
    ------------------------------------------------------------
    -- pt local for now -- file formal parameter not allowed with a public method
    procedure WriteBinName ( file f : text ; S : string ; Prefix : string := "%% " ) is
    ------------------------------------------------------------
      variable MessageCount : integer ;
      variable MessageIndex : integer := 1 ;
      variable buf : line ;
    begin
      MessageCount := CovMessageVar.GetCount ;
      if MessageCount = 0 then
        write(buf, Prefix & S & GetCovModelName) ; -- Print name when no message
        writeline(f, buf) ;
      else
        if CovNameVar.IsSet then
          -- Print Name if set
          write(buf, Prefix & S & CovNameVar.Get) ;
        elsif AlertLogIDVar /= OSVVM_ALERTLOG_ID then
          -- otherwise Print AlertLogName if it is set
          <b>write(buf, Prefix & S & GetAlertLogName(AlertLogIDVar));</b>
        else
          -- otherwise print the first line of the message
          MessageIndex := 2 ;
          write(buf, Prefix & S & string'(CovMessageVar.Get(1))) ;
        end if ;
        writeline(f, buf) ;
        for i in MessageIndex to MessageCount loop
          write(buf, Prefix & string'(CovMessageVar.Get(i))) ;
          writeline(f, buf) ;
        end loop ;
      end if ;
    end procedure WriteBinName ;
</pre>

Changing the line to 

`write(buf, Prefix & S & string'(GetAlertLogName(AlertLogIDVar)));`

solved the issue.

Best regards,
Michael Kiesenhofer

JimLewis, 2017-06-27:
OOPs.   Yep.  I don't test with GHDL and I forgot the GHDL'isms that do not exist in other tools.

Paebbels, 2017-07-02:
Hello,

why does OSVVM provide these two overloads?
* `impure function GetAlertLogName(AlertLogID : AlertLogIDType) return string` ([view in source](https://github.com/OSVVM/OSVVM/blob/master/AlertLogPkg.vhd?ts=2#L1016))
* `impure function GetAlertLogName(AlertLogID : AlertLogIDType := ALERTLOG_BASE_ID) return string` ([view in source](https://github.com/OSVVM/OSVVM/blob/master/AlertLogPkg.vhd?ts=2#L2787))

The latter one calls the former one to return the access types content.

I think it could be reduced to just one function. The user shouldn't see the change at all.

JimLewis, 2017-07-03:
The first GetAlertLogName is internal to the protected type - which is declared in the package body - and is never visible to the user.   It has direct access to the PT internals.

The 2nd GetAlertLogName is the user interface into the package.    Since the shared variable is declared internal to the package, it provides a calling interface to the PT methods of the shared variable.

closed 2017-07-03 by JimLewis

tmeissner, 2017-09-08:
I've stumbled over this issue yesterday when I tried to update OSVVM in a project of me. This issue isn't fixed until now, so I had to make the change @MichaelK92 suggests to make it work. I think, this should be fixed to allow GHDL users to use OSVVM without needing to patch.

JimLewis, 2017-09-08:
Hi Torsten,
I put an alternative patch my development branch.   Rather than add the annoying, GHDL specific type qualifier, I changed the AlertLogID to an integer type, rather than a subtype of integer.   This way the conflict at that point should go away since they are different types.   It is a big change as if someone is currently storing an AlertLogID in an integer rather than AlertLogIDType, their code will break.

I understand GHDL's view of the standard, OTOH, I have not seen another simulator behave this way.   

Jim

Paebbels, 2017-09-08:
So the fix is available since this commit: 3adfb8da64d10df18b5bd4489ec25e7a8296a423

bafeigum, 2018-03-09:
As of the latest rev, this issue still exists. Currently have to make the change MichaelK92 suggests.

Paebbels, 2018-03-09:
@bafeigum  
No the issue is fixed! See branch `dev` or commit 3adfb8da64d10df18b5bd4489ec25e7a8296a423.

Please see also my last comment in this thread.

JimLewis, 2018-03-09:
It is fixed in the Dev release.   This will be released into the main branch shortly.  We test with Mentor and Aldec tools.   Neither of which have this issue, so it is hard to track.   

I first tried to address this by making AlertLogIdType a separate type definition rather than a subtype.   This is a better fix as it makes it so we don't need to worry about future occurrences of this problem.  Unfortunately this caused problems in another vendor tool, so the current dev branch fix has the string' fix.   

Long term, AlertLogIdType will be a separate type and this will no longer be an issue.

tmeissner, 2018-03-09:
Maybe we should put a hint to the Dev Branch in the README. So users can test the latest features / fixes if they want.