OSVVM/OSVVM #25: Should the final alert report log unused AlertLogIDs?

Paebbels, 2018-02-22:
The final testbench reports all collected alerts in a testbench.

Should that report mark unused AlertLogIDs / LogNames in the report?

JimLewis, 2018-02-22:
Do you want them noted or not noted?   For "not noted", see ReportNonZeroAlerts instead of ReportAlerts

Paebbels, 2018-02-23:
I would like to see all lines, but giving me a note/hint that a certain ID was never used in the system. I then can remove that "dead" ID or review my testbench code.

JimLewis, 2018-02-23:
This would be like writing to a signal or variable, but never reading it.    

Sounds similar to what an editor like Sigasi can already do - to some degree since an id can be used in a separate design.

Otherwise it is expensive to add in the model.

JimLewis, 2018-02-23:
Do you have a critical requirement or use model for this.

Paebbels, 2018-02-23:
I was wondering, if OSVVM has one or more internal counters that indicate by being unchanged (being zero) that an ID was not used. E.g. the AlertLogPkg knows when no affirmation was triggered.

I'm asking not for a global "no affirmation", but for a per ID "no affirmation".

Use case:  
In a lab I saw a student generating multiple IDs, but he logged everything to the parent ID. So the created child IDs were useless. A hint like ChildID = 2 is never used, could help to identify such mistakes in any model.

JimLewis, 2020-05-09:
Some of my IDs are used specifically for certain classes of protocol checkers - such as the model never received a response within a given period and it timed out and failed.   

For most tests, this one will indeed have a 0 count.    Hence, I cannot make a conclusion on a zero count.   

Also while we have plans to count self-checking/affirmations, there is no plan to count calls to logs.   

These are found when you are verifying the model implementation and triggering all logs and alerts the model supports.   

Unfortunately we cannot assume that a 0 count is a bad thing.

closed 2020-05-09 by JimLewis