OSVVM/OSVVM #27: Allow Mirror on Specific Alert Levels

SkydiverTricky, 2018-06-25:
I currently have testbenches that use a transcript file. But I also like specific messages to also display on std.textio.OUTPUT. I do this in the testbench by turning on/off the mirror around specific messages, but for alerts inside the scoreboard I have no control over this. When an error/failure occurs and Error count is reached, ReportAlert occurs inside the scoreboard and cannot be mirrored to console.

I use the log to log everything - all scoreboard matches as well as errors. It would be good to have only the errors pop out to console.

SkydiverTricky, 2019-02-19:
As a note to this, when I run tests on jenkins, I have to have a post run TCL script to pull the ERRORs out of the log into the console to allow them to the be sent to the log parser to allow a job to be set to PASS or FAIL. 
As an aside, the ReportAlerts is called from within the AlertLogPackage when an AlertStopCount is reached, so I cannot mirror only this to console for quickly viewing the test status from the Jenkins console without my TCL script above.

This is the TCL I use (which can be quite slow with large OSVVM log files):

```
proc promote_errors { log_file } {

        set f [open $log_file r]
        set file_data [read $f]
        close $f

        set data [split $file_data "\n"]
        set printrest 0

        foreach line $data {
            if { [regexp {DONE} $line] } {set printrest 1 }
            if { [regexp {ERROR} $line] || [regexp {FAILURE} $line] || $printrest == 1 } { puts $line }
        }
    }
```

JimLewis, 2019-02-19:
I will put these on the update list.   Both seem interesting and definitely worthy to add, particularly the 2nd one.

JimLewis, 2020-05-09:
Resolution:
Allow mirroring for Log, Alert, and Done to be Separately