OSVVM/OSVVM #32: Add Check Function to Scoreboard

SkydiverTricky, 2019-02-15:
It would be useful to have a check function in the scoreboard that returns a boolean. This way you can easily connect a signal to the function to allow quick debugging on a waveform. I have the following current workaround:

err_cnt   := GetAlertCount(sboard.GetAlertLogID);
sboard.check(act);
missmatch <= (GetAlertCount(sboard.GetAlertLogID) /= err_cnt), false after 1 ns;

I think it would be easier to simply write:

missmatch <= sboard.check(act);

It would also allow a testbench to take user controlled action on a missmatch.

JimLewis, 2019-02-18:
Ok.  I will add this to my todo list.

If you are just trying to track when an error occurs, you might check out if your simulator supports displaying variables within a protected type.    WIthin the scoreboard package, the error count is kept in the 
   variable ErrCntVar : IntegerArrayPointerType := new IntegerArrayType'(1 => 0) ;

Similarly, if you want to track the overall error count for all errors, the AlertLogPkg on the DEV branch has a variable to do this:
variable ErrorCount : integer := 0 ;
variable AlertCount : AlertCountType := (0, 0, 0) ;

With these you should be able to use your simulator's jump to change capability to find the error location in the wave window.

Note that ErrorCount and AlertCount were added specifically for this functionality.   Note it is the the DEV branch and I missed adding the "Var" suffix that I normally have for the other PT body variables, so the name may change when it goes to the Main branch.

SkydiverTricky, 2019-02-19:
Thanks
My code lives inside a AXI bus sink BFM/checker entity, which is intended for use by users who may not be familiar with OSVVM or more advanced simulator/VHDL behaviour (most users have never seen a protected type before, let alone used one), so having a missmatch signal directly available on the waveform may be more obvious to them.

JimLewis, 2019-02-19:
Not hard to do.   Just tedious to add the additional tests to support the overloading.

JimLewis, 2020-05-09:
Resolution:  
Overload Check so that it returns boolean to indicate PASS/FAIL status of check

JimLewis, 2020-05-18:
Released in 2020.05

closed 2020-05-18 by JimLewis