OSVVM/OSVVM #33: Scoreboard: Add Default value to flush, or add new flush procedure to empty scoreboard

SkydiverTricky, 2019-02-19:
I would like to empty all expected data from a scoreboard. For example, when a reset occurs in the middle of a test, it might be appropriate to just empty all queues in the scoreboard and start again.

I previously used this, but it is not appropriate because it destroys all the configured IDs and any other config:
```
queue_reset_proc : process(axi_reset)
  begin
    if axi_reset = '1' then
      sboard.Deallocate;
      sboard.Initialize;
    end if;
  end process;
```

JimLewis, 2019-02-19:
Have you tried:
`sboard.Flush(sboard.GetItemCount) ;`

Easy to add, but given the above, do you still want it?   Indexed scoreboards keep their own item count, so those would also be simple.   Tagged scoreboards probably require a little more work and may warrant the overloading.

SkydiverTricky, 2019-02-22:
Makes sense. Ill get this closed.

closed 2019-02-22 by SkydiverTricky