OSVVM/OSVVM #34: Add getBinName function based on cover point

SkydiverTricky, 2019-02-22:
I have a test where I use randCovPoint to drive the input:
idx  := cov.randCovPoint;

I have named each of my bins. But I now have a cover point but I cannot get the name of the bin for reporting, something like this:

log("Sending Input " & to_string(idx) & " contained in Bin: " & cov.getBinName(idx));

Unless I make 1 bin per cover point, I cannot get the correct name of the bin. And this seems wasteful if I have to make several bins with the same name, and more complicated if I want to set the "AtLeast" values for specific bins.

maybe the function could be called getBinNameFromCovPoint

JimLewis, 2019-02-23:
This makes sense and goes with other plans - such as one of the GetBin requests

JimLewis, 2020-03-09:
I have been reviewing the issues and working on new planned features.   As I am doing that I have been reviewing these to make sure I get other requests in at the same time.

I noticed that GitBinName already exists.   It was added in:
--    03/2016   2016.03    Added GetBinName(Index) to retrieve a bin's name

This is what you were looking for right?   If yes, then I would like to close this.

JimLewis, 2020-03-09:
OK.   I see you are looking for a little more, but the information is still there.

When you do RandCovPoint, it logs the value it randomized in LastIndex
variable and you can retrieve it using, GetLastIndex.   

Hence, your getBinNameFromCovPoint should be just:
Cov.GetBinName(Cov.GetLastIndex) 

I am looking at further extending the capability to allow us to get then
index when we call ICover (as a return) and maybe have it save its
value either in LastIndex or in another internal variable.

If it is separated, then maybe we would have a GetLastCovIndex and GetLastRandIndex, however, I am not sure the value of separating them.

JimLewis, 2020-05-09:
Enhancement, separate from OP issue:
Record bin found by ICover in LastIndex so it can be retrieved by GetLastIndex.  
Furthermore, consider separating LastIndex into LastCovIndex and LastRandIndex, however, that is probably not needed.

Also note that in addition to RandCovPoint, there is also RandCovBinVal if you want to do other fun stuff.

JimLewis, 2020-05-09:
Cov.GetBinName(Cov.GetLastIndex) will return the bin name after either RandCovPoint or ICover.

I did not see a need to have GetLastCovIndex and GetLastRandIndex, so if for some reason you need to differentiate, you will need to request it.

JimLewis, 2020-05-18:
Released in 2020.05

SkydiverTricky, 2021-06-07:
Hi @JimLewis 
Just want to say you may want to add something about the below in the docs, as I was just trying to add it to my TB and I forgot all about this and was confused as to why random values were returning very odd bin names and then get an out of range (1 to 128) error. Without the below, I forgot the bin indexes are unrelated to the coverage ranges.

> Cov.GetBinName(Cov.GetLastIndex) will return the bin name after either RandCovPoint or ICover.
> 
> I did not see a need to have GetLastCovIndex and GetLastRandIndex, so if for some reason you need to differentiate, you will need to request it.

JimLewis, 2021-06-07:
Add documentation for the above

JimLewis, 2022-01-31:
Checked doc and verified GetBinName is there.   It is.  Closing issue.

closed 2022-01-31 by JimLewis