OSVVM/OSVVM #35: Position of Simulation Time

SebDunst, 2019-02-25:
Hello Jim,
I am a QuestaSim user and I have an issue regarding the position of the simulation time when doing logging.
The problem occurs is if I want to log a time value e.g.: "Pulse length = 20 ns at 123450000 ns".
The link between the Transcript window and the Wave window is then not working anymore (clicking with the mouse in Transcript window onto the simulation time normally sets the cursor in the Wave window). I made a support case at Mentor and they said that this link in looking for the first timing value in each line.
So if the simulation time would be printed out not at the end of the line this problem could be solved.

Best regards,
Sebastian

JimLewis, 2019-02-25:
Hi Sebastian,
The problem with putting time first is that then you have to introduce some sort of arbitrary justification to the time value - otherwise, the output shifts for each power of 10 in time.   

If there is enough interest,  it is possible to make the location of where time prints configurable.   I hate configuration options though as they mean more testing and add little value.   

For the mean time, it is fairly easy to copy the time value and paste it into the wave window to get it there.

Jim

tmeissner, 2019-02-25:
Wow. I also use Modelsim / Questa at work and didn't know, that you can click on a timestamp in the transcript window to set the cursor in the wave window. That is something I always miss when using OSVVM Logs or Alerts. Sometimes I used plain VHDL asserts because of this, as they can be added to the wave window.

JimLewis, 2019-02-25:
Hi Torsten,
Do you also have issues in ModelSim now?   Sebastian only reported issues when putting a time value in the message as ModelSim seems to only understand the first value in the line.

Best Regards,
Jim

tmeissner, 2019-02-25:
Ah, okay. You mean then you have 2 timestamps in one line, and Modelsim / Questa only recognizes the firestone, which isn't the time at of the event. I never had such a log message.

@SebDunst Is it possible (when no second time in the log message) to clock on the line and set the cursor in the wave window with that? I didn't know that. I have to try that tomorrow. I always copy the timestamp from the log message and use that in the wave window to set the cursor ... Simply clicking on it would be a far more smarter way :)

SebDunst, 2019-02-27:
@tmeissner I am doing this all the time (when debugging my tb)
@JimLewis I see the problem with too many configuration options. An idea could be, that the simtime is moving (fix) somewhere before or after the LogID name. And then the user is able to set the column width for the simtime and for the LogID name.

JimLewis, 2019-02-27:
Hi Sebatian
That is a non backward compatible change to fix a broken vendor (OSVVM perspective) feature.  That would be bad for current users.

If accepted it would have to be an option.  

OTOH, You can print a second message such as a log at the same time step that does not have a time value.   You could also concatenate a new line so you message prints your time on the first line on the OSVVM time on a separate line

JimLewis, 2022-09-12:
@SebDunst @tmeissner 
This was updated some time this summer.   
Try out the options:
```
    SetAlertLogOptions(WriteTimeLast => FALSE) ; 
    SetAlertLogOptions(TimeJustifyAmount => 15) ; 
    SetAlertLogJustify ; 
```

closed 2022-09-12 by JimLewis