OSVVM/OSVVM #37: Scoreboard: Add GetCurrentItemCount function

SebDunst, 2019-03-18:
Hi Jim,
I often use the Scoreboard "only" as a Fifo. I can push values into it, and I can pop values from it. What I am missing is a function, which returns the current number of values stored in the Fifo, not the overall values ever have been stored (GetItemCount).
Something like "GetCurrentItemCount()" or "GetStoredItemsCount()".

What would you think about this?
Regards,
Sebastian

SebDunst, 2020-01-16:
Hi Jim,
here would be my solution for this item. The file "ScoreboardGenericPkg_SebDunst.txt" is attached at this post. The new function is called "GetCurrentItemCount()".
Regards,
Sebastian
[ScoreboardGenericPkg_SebDunst.txt](https://github.com/OSVVM/OSVVM/files/4070267/ScoreboardGenericPkg_SebDunst.txt)

JimLewis, 2020-01-22:
Hi Sebastian,
This should be an easy add.    For now I will write the code.   In the future, we are moving to IEEE and will be having a CLA to allow users to be able to contribute code.

We may need to update the naming convention to disambiguate.  
What I see we currently have is:
GetItemCount - returns the push count
GetCheckCount - returns number of items checked
GetDropCount - returns number of items dropped.

The word "Current" to me is ambiguous.   How do I intuitive differentiate
it from GetItemCount and GetCurrentItemCount?  
What you are asking for is the number of words in the FIFO.
If we keep short names, like GetCheckCount, then perhaps what you are 
asking for should be GetFifoCount.   Maybe it also means that GetItemCount
should be changed to GetPushCount (GetItemCount would maintain its
current functionality - would use an alias but not allowed until 2019).

Of course, Item could come before each of those in the name.   

With that then maybe we should have a GetPopCount, but then what
value does it return - especially since check calls pop.  Pop count could
count the number of calls to just pop or pop and check or 
pop, check, flush.

Thoughts?

Best Regards,
Jim

SebDunst, 2020-01-23:
Hi Jim,
the name "GetFifoCount" seems to me the obvious right solution.
Also replacing the name "GetItemCount" with "GetPushCount" would clarify the interface.

Another idea would be to create a new (and very simple) package called "fifo" or "queue".
If I want a scoreboard with all that checking functionality, I will go with the existing scoreboard. But if I only need a dynamic fifo or queue (Push, Pop, IsEmpty, GetFifoCount, Flush), then I would go with this new package.

Best regards,
Sebastian

JimLewis, 2020-01-23:
Hi Sebastian,
I thought about refactoring the FIFO, however, I could not see what I gain - and it certainly amounts to another package to test and maintain.

Best Regards,
Jim

JimLewis, 2020-05-18:
Released in 2020.05

closed 2020-05-18 by JimLewis