OSVVM/OSVVM #38: IncAlertCount Procedure should call stop with exit code 1

SkydiverTricky, 2019-05-26:
Shouldnt  procedure IncAlertCount () exit with code 1? 
The alert procedure exits with code 1.
This way CI can know if a test has failed without having to scrub the log?

Or maybe have a config item to set the exit code?

m-kru, 2019-09-06:
I also think that exit status code should not be 0 in such case.

JimLewis, 2020-03-09:
Planned for next revision

JimLewis, 2020-05-09:
I should clariify, AlertLogPkg will call stop with the current ErrorCount:    
std.env.stop(ErrorCount)

ErrorCount is a global variable in the AlertLogPkg protected type.  In some simulators it can be traced.   It can also be optionally be printed following Alert and before the Alert Level

JimLewis, 2020-05-18:
Released in 2020.05

closed 2020-05-18 by JimLewis