OSVVM/OSVVM #40: Add add to_boolean and to_std_logic to CoveragePkg.vhd.

rpaley613, 2019-11-20:
Please add the mirror of 
function to_integer ( B : boolean ) return integer ;
and
function to_integer ( SL : std_logic ) return integer ;

something like:

function to_boolean ( I : integer ) return boolean;
and 
function to_std_logic ( I : integer ) return std_logic;

regards,
Robert (Reuven) Paley

JimLewis, 2020-05-14:
What should be done for integer values that are not 0 or 1.
In to_std_logic, if not 0 or 1, return 'X' and issue an alert level error
In to_boolean, if not 0 or 1, return FALSE and issue an alert level error

Sound OK?   

Plan is that this will be in the 2020.05 release.   Testing now.

rpaley613, 2020-05-15:
Hi,

What is the goal of these functions, to simply convert logical 0 and 1? If
so, then I propose the to_std_logic should return 'U' and not 'X'. This
would better return the concept of "undefined" rather than "contention"

Regards
Robert (Reuven) paley

JimLewis, 2020-05-15:
Hi Rob,
I am not sure which is best.   'U' is actually Uninitialized formally in std_logic_1164.
Hence, no active effective drivers yet.    'X' is simply unknown.   If you consider
the resolution function resolved, 'U' is strongest - 'U' and anything is 'U'.

Of course, I started with MVL7 where there was only 1 'X','-','U' value.

Best Regards,
Jim

JimLewis, 2020-05-18:
Released in 2020.05

closed 2020-05-18 by JimLewis

rpaley613, 2020-05-19:
Thanks Jim. I need to upgrade from version 2017