OSVVM/OSVVM #46: Error on AlertLogPkg.vhd when using ModelSim 10.2c

suoto, 2020-03-07:
Been getting the following errors with OSVVM

    ** Error: osvvm/AlertLogPkg.vhd(984): No feasible entries for subprogram "PrintTopAlerts".
    ** Error: osvvm/AlertLogPkg.vhd(991): No feasible entries for subprogram "PrintTopAlerts".

The procedure's is defined as

https://github.com/OSVVM/OSVVM/blob/185a11d7bbf896e5fb66035cbad7fcb5ab505c79/AlertLogPkg.vhd#L881-L887

Errors occur here

https://github.com/OSVVM/OSVVM/blob/185a11d7bbf896e5fb66035cbad7fcb5ab505c79/AlertLogPkg.vhd#L978-L991

The code *is* correct -- this seems to be a bug on ModelSim itself and it has been fixed at some point between 10.2c and 10.7c (versions I have available for testing)

I did a bit of digging and it seems ModelSim is making confusion with the `AlertCount` identifier, which is used to name parameters in procedures/functions, constants, variables and record components. Either replacing the parameter for a different identifier (and adjusting uses) *or* calling `PrintTopAlerts` with positional arguments (no named parameters) seems to fix the problem.

I know it's a ModelSim bug from an old version, but would be great if OSVVM could work around it!

JimLewis, 2020-03-08:
Hi Andre,
Is there a reason you need to keep using such an older version?  Is it part of the current distribution of any FPGA tool?

I should also note that there are some bugs in 10.4 that I do not think were fixed until 10.4d or 10.4e.   I don't even think there was a work around in these.   

I understand the simple work around you proposed in the other post, but the coding style that activated the bug in ModelSim is a preferred coding style and is prevalent in other parts of OSVVM.   

In OSVVM, we work hard to make sure we keep working with the current versions of tools and actively modify code to work around their issues.   However, I don't think we can afford to make that a practice for really old revisions - unless there is a compelling reason.   

Best Regards,
Jim

suoto, 2020-03-09:
I understand that, no worries. This was just the version I got installed here and since the "fix" is very small I thought it might be worth suggesting. And in all honesty, I did not realize that 10.2c was actually 7 years old.

JimLewis, 2020-03-09:
Keep the thoughts coming though.   I am particularly interested in making sure that we work around issues in the more current releases.

closed 2020-03-09 by JimLewis