OSVVM/OSVVM #52: Scoreboard: Allow user to override the Received: XXX Expected: XXXX log message(s)

SkydiverTricky, 2020-07-22:
The current log message can be fairly difficult to decode if there are complex/long types used as the expected/actual types. Here is an example using AXIS types:

`KERNEL: %% Alert ERROR   in data_checker_inst,    Received: tid: 0xXX tdata: 0x2F2E2D2C2B2A29282726252423222120 tuser: 0x1 tdest: 0xX tkeep: 0xFFFF tstrb: 0xXXXX tlast: 0    Expected: tid: 0xXX tdata: 0x0F0E0D0C0B0A09080706050403020100 tuser: 0x1 tdest: 0xX tkeep: 0xFFFF tstrb: 0xXXXX tlast: 0    Item Number: 1 at 200 ns`

I have a custom report function that outputs much more detailed comparison inside the `match` function. This currently gives me the following output:

```
KERNEL: %% Log   ALWAYS  in testbench_tools_pkg, act: tid: 0xXX tdata: 0x2F2E2D2C2B2A29282726252423222120 tuser: 0x1 tdest: 0xX tkeep: 0xFFFF tstrb: 0xXXXX tlast: 0  at 200 ns
KERNEL: %% Log   ALWAYS  in testbench_tools_pkg, exp: tid: 0xXX tdata: 0x0F0E0D0C0B0A09080706050403020100 tuser: 0x1 tdest: 0xX tkeep: 0xFFFF tstrb: 0xXXXX tlast: 0  at 200 ns
KERNEL: %% Log   ALWAYS  in testbench_tools_pkg,                         X X X X X X X X X X X X X X X X                                                              at 200 ns
KERNEL: %% Alert ERROR   in data_checker_inst,    Received: tid: 0xXX tdata: 0x2F2E2D2C2B2A29282726252423222120 tuser: 0x1 tdest: 0xX tkeep: 0xFFFF tstrb: 0xXXXX tlast: 0    Expected: tid: 0xXX tdata: 0x0F0E0D0C0B0A09080706050403020100 tuser: 0x1 tdest: 0xX tkeep: 0xFFFF tstrb: 0xXXXX tlast: 0    Item Number: 1 at 200 ns
```

Ideally, I could just replace the default report with my own.

JimLewis, 2020-07-22:
For now, you can do:
SetReportMode(REPORT_NONE);

It is going to yack at you and say that that option is deprecated.   You can comment that alert out in the package.   It is going to do exactly what you wanted in the first place.

JimLewis, 2020-07-22:
There have been numerous issues related to "I want to print it my way".   I have plans for an update that will allow that.

SkydiverTricky, 2020-07-22:
Actually - I did consider it could be passed in as a generic, but that requires the existing generics, which would require a default generic package, and ActiveHDL has only offered packages as generics since v11.1.

SkydiverTricky, 2020-07-22:
Will be much easier with Generics on protected types  :)

JimLewis, 2020-07-22:
Yes will be much easier with Generics on PT.    Make sure to submit a bug report against it if your simulation vendor has not implemented this yet.

JimLewis, 2022-01-31:
Just confirming, this is still planned.   I am thinking by summer 2022.

Paebbels, 2022-01-31:
@JimLewis should we define 2-3 labels like:
* Next release
* Next 2-3 months
* Next 4-6 months

JimLewis, 2022-02-06:
@SkydiverTricky 
ScoreboardGenericPkg was created before we had the AlertLogPkg.   Since we have a good way of reporting errors with the AlertLogPkg, instead of a custom Match function that does the printing, you could do push on the sending side, and then pop on the checking side.   That will give you the expected value, from there you can compare it how you see fit.

JimLewis, 2025-02-13:
Now that vendors support overloading pop as a function (actually this has been true for some time now), the very similar to a check function is:

```vhdl
MyAffirmIf(ModelID,  ReceivedValue,  Pop(ScoreboardID), "My message ...") ; 
```

There are still plans to update the scoreboard to gain more flexibility, however, there is a reasonable solution today.    I also have to trade off this solution to the use model for the updates - if they are not better, this may become the primary solution.

JimLewis, 2026-09-10:
Periodically I do an issue review.   

I have reviewed planned updates for ScoreboardGenericPkg a number of times.   To give someone the total flexibility requires passing of the Affirmation.   Lets say we do that.   That would mean the use model is:

```vhdl
Get(StreamRec,  ReceivedValue) ; 
Check(ScoreboardID, ReceivedValue, "Message") ;
```

OTOH, lets say we strip the scoreboard down to being just a FIFO.    Then the use model would be as suggested above:
```vhdl 
Get(StreamRec,  ReceivedValue) ; 
MyAffirmIfEqual(CheckID, ReceivedValue, Pop(ScoreboardID), "Message") ;
```

2026.08 was just released.   One of the things it supports is multi-lined messages in Alerts, log, and hence affirmations.
For your type of messaging, the internal AffirmIF would look like:

```vhdl
AffirmIf(CheckID, DoTheyMatch(ReceivedValue, ExpectedValue), 
             to_string(ReceivedValue), 
             LF & to_string(ExpectedValue) & 
             LF & GenerateMixmatchXs((ReceivedValue, ExpectedValue)) ;
```

The LF in the text is now supported.   On an error, it will print as:
```
%%         200.0 ns  Alert ERROR   in data_checker_inst,   Received: tid: 0xXX tdata: 0x2F2E2D2C2B2A29282726252423222120 tuser: 0x1 tdest: 0xX tkeep: 0xFFFF tstrb: 0xXXXX tlast: 0
%%>                                                        Expected: tid: 0xXX tdata: 0x0F0E0D0C0B0A09080706050403020100 tuser: 0x1 tdest: 0xX tkeep: 0xFFFF tstrb: 0xXXXX tlast: 0  at 200 ns
%%>                                                                                     X X X X X X X X X X X X X X X X  
```

The scripts that do the highlighting will color all the lines of the Alert ERROR red.   It uses the %%> as a continuation indiation.

Alternately you can also do an indented line by adding an LF to the beginning of the message:
```vhdl
AffirmIf(CheckID, DoTheyMatch(ReceivedValue, ExpectedValue), 
             LF & to_string(ReceivedValue), 
             LF & to_string(ExpectedValue) & 
             LF & GenerateMixmatchXs((ReceivedValue, ExpectedValue)) ;
```

This will then print as:
```
%%         200.0 ns  Alert ERROR   in data_checker_inst,   
%%>                  Received: tid: 0xXX tdata: 0x2F2E2D2C2B2A29282726252423222120 tuser: 0x1 tdest: 0xX tkeep: 0xFFFF tstrb: 0xXXXX tlast: 0
%%>                  Expected: tid: 0xXX tdata: 0x0F0E0D0C0B0A09080706050403020100 tuser: 0x1 tdest: 0xX tkeep: 0xFFFF tstrb: 0xXXXX tlast: 0  at 200 ns
%%>                                               X X X X X X X X X X X X X X X X  
```

The downside of this is that LF is that it activates the WRAP feature (new with 2026.08).   If number of characters in your line nominally exceeds 162, then it will add additional LF which is perhaps not desirable.   As a new feature, this could be adjusted so that LF does not activate WRAP, but only '<' does.

If we compare a FifoGenericPkg vs a ScoreboardGenericPkg that accepts AffirmIfEqual on its interface, the net result from the use model is not much different.   It boils down to:

`Check(ScoreboardID, ReceivedValue, "Message") ;  -- updated scoreboard`

vs 

`MyAffirmIfEqual(CheckID, ReceivedValue, Pop(ScoreboardID), "Message") ; -- a FIFO`

If we reformulated the affirmation some so that used the ScoreboardID instead of the CheckID and it did the Pop internally, then the FIFO use model becomes just as simple as the Scoreboard use model.

`MyAffirmIfEqual(ScoreboardID, ReceivedValue, "Message") ;  -- FIFO`

JimLewis, 2026-09-10:
All that said, I think passing AffirmIfEqual to the scoreboard is a better long term interface for the Scoreboard, so I expect the change to happen, just not sure when.    Also I am not sure if it will necessarily be better than a FIFO.   And a FIFO you have today.

There are also plans to re-factor to separate the PT interface from the Singleton interface.   It may be that the PT interface does not get the update.