OSVVM/OSVVM #54: Can OSVVM offer also a decrement procedure in TbUtilPkg?

Paebbels, 2020-09-25:
Currently OSVVM offers a TbUtilPkg with an `Increment` and `WaitFOrToggle` procedures.

Currently no `Decrement` is offered. Is it possible to add this missing feature for completeness?

Current API for integer value passing based on toggle events:
```vhdl
  -- Integer type versions
  procedure Increment ( signal Sig : InOut integer ; constant RollOverValue : in integer := 0) ;
  procedure WaitForToggle ( signal Sig : In integer ) ;
```

JimLewis, 2020-09-25:
Yes.  Do you have a use for it?   Completeness is good and will go on the "some day list".   Items with use models get priority.

Currently I am using Increment for handshaking and transaction tracking - which always add 1.

Paebbels, 2020-09-25:
In some cases you might want to count down instead of count up. Also incrementing by N might be required by some usecases. It could be solved with an optional parameter and default value `1`. Atleast, this could then be used to decrement when setting the incValue to `-1`.

In OSVVM models I saw, integers are incremented on both sides of a transaction FIFO, so:
* it's increments on both sides (producer and consumer)
* fillstate = difference of both counters
* it's only one driving process, hence no value resolution or protected type needed.
  * A PT would shift the problem again, because integer counts are used to send PT instance data updates from producer to consumer

So maybe we are both looking for a `'event` on protected type instances...

JimLewis, 2020-09-26:
Maybe what we need is an attribute of a shared variable and a method of a shared variable that creates an equivalent signal that toggles when either any method of the shared variable or share variable method is called.   

Something like: 
share variable SB : ScoreboardPType_slv ; 

process
...
   while SB.Empty loop 
      wait until SB.Push'transaction ; 
  end loop ;

However, to facilitate modularization, we need to access these parameters in subprograms.
Maybe what we need to do is have transformation rules:
  1:  A signal declared in a process, is equivalent to a signal declared in the containing declarative region that is only visible to that process.
  2: A signal declared in a subprogram is equivalent to a signal declared in the containing declarative region that is only visible to that subprogram.   

For nested subprograms and subprograms declared in a sequential declarative region, rules 1 and 2 are repeated until rule 1 is applied.

Maybe we should move this discussion to the WG issues list.