OSVVM/OSVVM #57: Add function to check if a value is in a specific bin

rhead-12345, 2021-05-24:
I have the following Bins, created as objects to give a simple reference:

```vhdl
    constant BIN_RUNT_FRAMES         : CovBinType := GenBin(16,                         63,                           1);    -- Short frames
    constant BIN_SHORT_FRAMES        : CovBinType := GenBin(64,                         79                             );    -- Shortest frames, covering all word width positions (16 bytes per word)
    constant BIN_STANDARD_FRAMES     : CovBinType := GenBin(80,                         ETHERNET_FRAME_SIZE_MAX-16-1, 1);    -- All standard Ethernet lengths
    constant BIN_LARGE_FRAMES        : CovBinType := GenBin(ETHERNET_FRAME_SIZE_MAX-16, ETHERNET_FRAME_SIZE_MAX        );    -- Longest standard ethernet lengths, covering all word width positions (16 bytes per word)
    constant BIN_JUMBO_FRAMES        : CovBinType := GenBin(ETHERNET_FRAME_SIZE_MAX+1,  9000,                         1);    -- Jumbo Frames

...

-- add the coverage
cov.AddBins("Runt Frames    ", G_N_HITS_PER_COV_PT, BIN_RUNT_FRAMES    );
cov.AddBins("Short Frames   ", G_N_HITS_PER_COV_PT, BIN_SHORT_FRAMES   );
cov.AddBins("Standard Frames", G_N_HITS_PER_COV_PT, BIN_STANDARD_FRAMES);
cov.AddBins("Large Frames   ", G_N_HITS_PER_COV_PT, BIN_LARGE_FRAMES   );
cov.AddBins("Jumbo Frames   ", G_N_HITS_PER_COV_PT, BIN_JUMBO_FRAMES   );
```

When I get the coverage, I cant work out if there is a simple method to check if a value returned from `cov.randCovPoint` is in a specific Bin, other than checking the ranges of all the bins. something like

```vhdl
testPoint := cov.randCovPoint;

if isInBin(testPoint, BIN_RUNT_FRAMES) then
  -- Generate Runt Frame
elsif isInBin(testPoint, BIN_SHORT_FRAMES then
  -- generate short frame

--etc
```

Currently, I can work around this by matching a testPoint to a bin name. 
Maybe the name labels could also be added to the CovBinType as an unconstrained string?

JimLewis, 2021-05-24:
Thanks for submitting issues with use cases.

A superset of this has been in the planning for some time.   My plan was to do it against the coverage model `cov` and not against objects of CovBinType.   Maybe we need a bin name and a group name - OTOH, maybe group name should be done like case 2 below. 

There will be an update this summer.   Working towards being able to create a UCIS - just like SV.   After that point, we will be able to do something like this.   

For the mean time, I have two ideas for you if you are not already doing them.
1) From where you are at now:
```vhdl
testPoint := cov.randCovPoint ;
testName := cov.GetBinName(cov.GetLastIndex) ;  -- Last Index randomly generated

case testName is    -- you conveniently made them all the same length
  when "Runt Frames    "   => 
    -- Generate Runt Frame 

  when "Short Frames   "   => 
    -- Generate Short Frame 
```

2)  Add frame type information to your coverage model.   
```vhdl
type FrameType is (RUNT, SHORT, STANDARD, LARGE, JUMBO) ; 

cov.AddCross("Runt Frames    ", G_N_HITS_PER_COV_PT, FrameType'Pos(RUNT),     BIN_RUNT_FRAMES    );
cov.AddCross("Short Frames   ", G_N_HITS_PER_COV_PT, FrameType'Pos(SHORT),    BIN_SHORT_FRAMES   );
cov.AddCross("Standard Frames", G_N_HITS_PER_COV_PT, FrameType'Pos(STANDARD), BIN_STANDARD_FRAMES);
cov.AddCross("Large Frames   ", G_N_HITS_PER_COV_PT, FrameType'Pos(LARGE),    BIN_LARGE_FRAMES   );
cov.AddCross("Jumbo Frames   ", G_N_HITS_PER_COV_PT, FrameType'Pos(JUMBO),    BIN_JUMBO_FRAMES   );

(Frame_Int, testPoint) := cov.randCovPoint ;

case FrameType'Val(Frame_Int) is    -- you conveniently made them all the same length
  when RUNT   => 
    -- Generate Runt Frame 

  when SHORT   => 
    -- Generate Short Frame 
```

rhead-12345, 2021-05-24:
Thanks @JimLewis 
Something along the lines of the first had occured to me (making them all the same length looks nicer in `writeBin`)
The second is interesting - I might give that a go.

JimLewis, 2021-06-09:
As I review this again, of course isInBin is a possible and perhaps even easy thing to write, however, its expense will increase as the number of individual bins in each item increase.   So searching bins like BIN_RUNT_FRAMES is going to be inexpensive since there is only one bin, but searching bins like  BIN_SHORT_FRAMES and BIN_LARGE_FRAMES is going to be similar to 1 above.   

Solution 2 above is going to be a much more efficient solution.   

If someone writes an isInBin, I would accept a pull request for it, but based on priority (I would rank this item low compared to other work), I don't think I will have the time in the next 3 months at least.

JimLewis, 2025-02-15:
I reviewed this again in preparation for the 2025.02 release.   

What I see in the issue is that the request is for support to introspect a constant of type CovBinType.   CovBinType is an input that is used to construct the internal data structure.  It would be simple enough to accomplish this.   However as complexity grows - as it does with BIN_SHORT_FRAMES and BIN_LARGE_FRAMES the cost of the search will increase.   Going further, this is just the start of something like this as one can create an entire cross product in a constant.   This would again would be easy enough to implement, but would take time to search.

In the proposed approach 2 above, we see that OSVVM can already solve this problem, by having the user add a discriminator to the coverage model.   This solves removes the search issue down and adds a minimal amount of information to the coverage data structure - two integer values per bin in the coverage model - where this coverage model has 39 bins (assuming I counted right).   Not a bad trade-off.

In general, CoveragePkg intends to provide introspection on the internal coverage data structure.   

I would still be willing to accept a pull request that does the search of an object of type CovBinType, but since OSVVM already provides what I suspect is a better solution, I am going to close this issue.

closed 2025-02-15 by JimLewis