OSVVM/OSVVM #58: VHDL 2019 Suggestion - Add function to link CovPType Random generation to a specified RandomPType object

SkydiverTricky, 2021-06-09:
Sorry if you already have plans for this.
Currently, I have In house verification IP entities which have RandomPTypes in them. The seed is passed in via a generic to the testbench so I can set the seed from the CI_JOB_ID system variable to get a new pattern on each run in gitlab CI, but still allow me to recreate the pattern in case of an error. Because this is simply an integer, to ensure each has a unique random pattern, I pass `G_RANDOM_SEED` as a generic to the testbench, the for each VIP instatiation, I do the following generic maps:

```
G_RANDOM_SEED => (G_RANDOM_SEED + 1),
....
G_RANDOM_SEED => (G_RANDOM_SEED + 2),
...
G_RANDOM_SEED => (G_RANDOM_SEED + 3),
...
cov.InitSeed(G_RANDOM_SEED+4);
--etc
```

With VHDL 2019, I can modify the VIP entities to have a randomPType shared variable in the port map so I can simply use a single randomPType throughout the testbench. Ideally I would have a mechanism to pass the same object into the coverage object(s) so I can simply call InitSeed once.

SkydiverTricky, 2021-06-09:
One suggestion : maybe the `randInt` function can be passed in a generic to the `CovPtype`, which by default is connected to a randomPType object. This way the user could simply map it to the randInt function on their randomPType object in their testbench.

```vhdl
package coveragePkg is
  
  shared variable rand : randomPType;
  
  
  type CovPType is protected
    generic (
      function randInt return integer is rand.randInt;
    );
  ...
  end protected;
  
end package coveragePkg;

-- User testbench

shared variable randTB : randomPType;

shared variable cov : CovPType( randInt => randTB.randInt);
```

JimLewis, 2021-06-09:
First, don't ever be sorry for throwing out suggestions - whether they are in the plans or incorrect - they have potential to lead us to something that is good.

What I think you are asking is, to simplify simulation setup you would like one randomization object from which to do all randomization in a simulation.   You can do this with VHDL-2002 by putting your randomization object in a package.   However, that will give you the same situation Verilog had and that led to the creation of SystemVerilog as an attempt to fix it.   So don't do this until you read on.

Generally, I run my simulations in the same way the hardware runs - with many independent processes (threads) interacting with each other - if you are randomizing in your verification components, then you are too.    If I were then to share a randomization object (and hence, the seed) between two different processes  and the processes randomized during the same delta cycle, then the results of the randomization would depend on the order the processes run in.   Unfortunately order of process execution can change between optimizations and between simulators.   Hence, if you find a bug, fix the bug, you recompile - and reoptimize the design.   At this point, you are not guaranteed that the processes will run in the same order and the test case generated can be different.

Going further, if you want uniformity, sometimes a thread will need multiple randomization objects.   For example, if I randomize packet header information, then randomly select a packet length, and then randomly generate the packet.    Hence, each execution of the randomization thread will do N + PacketLength number of randomizations.   Randomization theory tells us that for a given uniformly distributed randomization object, every Nth pull form it will also be uniformly distributed.   So as long as our randomization process does the same number of randomizations each time, it will be uniformly distributed, but if you randomization the packet contents (and the packet length is also random), then it will not be uniformly distributed - which simply means that the packet contents needs a separate randomization object.   

With this in mind, I like your current method better.    So what don't you like about it?   If there is something, I can probably help find a solution.   

Currently I am working on data structure stuff.   This week, I am working on RandomPkg.   There will be some interesting new capability coming up.  If I understand what is going wrong for you, I may be able to make some adjustments to what I am doing to accommodate something.

SkydiverTricky, 2021-06-09:
Yes, the process order problem hadnt occured to me but makes perfect sense and why what I suggest isnt a good idea. I was only suggesting it so I dont need all these initialisations. With the current method re-creating any failed runs is simply a case of providing the same seed and it works well. I guess as they say - aint broke dont fix..

JimLewis, 2021-06-09:
Once you get it setup though, there is only one thing to change though right?   The top level input.   The rest of them generate off of that ok right?

JimLewis, 2021-06-09:
One of the issues with the current seed approach is that when the seeds are close together numerically, they seem to generate the same first number - the remaining ones diverge though.   So one change that is being made is after the seed is generated, we are doing a randomization to churn the seeds.

SkydiverTricky, 2021-06-10:
Yes, like you say, once its started it runs fine. All I do is post in the data and let it flow...

JimLewis, 2021-09-01:
New seed approach in 2020.07 removes concerns about seeds.   When called as a PT, all functions use the old seed methods.   To get the new seed methods you either need to ask for them (see doc on randompkg) or use the new interface to coveragepkg.

WIth that, I don't see any reason to keep this open, so I am closing it.

closed 2021-09-01 by JimLewis