OSVVM/OSVVM #59: Alerts - Option to print entire path to alert name.

SkydiverTricky, 2021-06-29:
I cant find if the following is possible - i would like to print multiple levels of the heirarchy name on and alert/log. Currently it only prints the name directly associated with the ID. Thats usually fine, but I have a verification component with multiple IDs inside that is instantiated twice in the testbench - so I cant easy tell which one is generating them:

```
KERNEL: %% Log   ALWAYS  in Default, Pushing write transaction to interface 0 at 750 ns
KERNEL: %% Log   ALWAYS  in Default, Pushing write transaction to interface 1 at 750 ns
KERNEL: %% Log   PASSED  in AWADDR Checker,    Received: axid: 0x00 axaddr: 0x00000000 axlen: 0x06 axsize: 100 axburst: 01    Item Number: 1 at 780 ns
KERNEL: %% Log   PASSED  in WDATA Checker,    Received: wdata: 0x00000006000000640000000026F2CDBF wstrb: 0xFFFF wlast: 0    Item Number: 1 at 780 ns
KERNEL: %% Alert FAILURE in AWADDR Checker, Valid transaction on Interface with no transaction expected at 780 ns
KERNEL: 
KERNEL: %% Alert Stop Count on FAILURE reached in AWADDR Checker at 780 ns 
KERNEL: %% DONE  FAILED  debug  Total Error(s) = 1  Failures: 1  Errors: 0  Warnings: 0  Passed: 10  Affirmations Checked: 10  at 780 ns
KERNEL: %%    Default                     Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%    OSVVM                       Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%    testbench_tools_pkg         Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%    axi_sim_utils_pkg           Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%    pcap_top_tb                 Failures: 1  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%      AXI4L Regs Driver         Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        AWADDR Driver           Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        WDATA Driver            Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        BRESP Checker           Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        ARADDR Driver           Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        RDATA Checker           Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%      RAM Interface 0           Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        AWADDR Checker          Failures: 0  Errors: 0  Warnings: 0  Passed: 1
KERNEL: %%        WDATA Checker           Failures: 0  Errors: 0  Warnings: 0  Passed: 1
KERNEL: %%        BRESP Driver            Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        ARADDR Checker          Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        RDATA Driver            Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%      RAM Interface 1           Failures: 1  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        AWADDR Checker          Failures: 1  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        WDATA Checker           Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        BRESP Driver            Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        ARADDR Checker          Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%        RDATA Driver            Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%      Register Scoreboard       Failures: 0  Errors: 0  Warnings: 0  Passed: 8
KERNEL: %%      Packet Coverage 0         Failures: 0  Errors: 0  Warnings: 0  Passed: 0
KERNEL: %%      Packet Coverage 1         Failures: 0  Errors: 0  Warnings: 0  Passed: 0
```

So with the messages coming from AWADDR/WDATA checker, I cant tell which RAM interface is reporting the messages. Ideally I could just specify the number of layers of heirarchy to report rather than just all of it.

JimLewis, 2021-06-29:
What I use is:
  constant MODEL_INSTANCE_NAME : string :=
    IfElse(MODEL_ID_NAME /= "", MODEL_ID_NAME, PathTail(to_lower(Axi4Master'PATH_NAME))) ;

Here MODEL_ID_NAME is a generic which is initialized to "".

Then when setting the IDS, I do:
```vhdl
Initialize : process
  variable ID : AlertLogIDType ;
begin
  ID                      := GetAlertLogID(MODEL_INSTANCE_NAME) ;
  ModelID                 <= ID ;
  ProtocolID              <= GetAlertLogID(MODEL_INSTANCE_NAME & ": Protocol Error", ID ) ;
```

JimLewis, 2022-01-31:
Quick update, this is on my list to solve.   

What are your expectations?   
   * Should it be a global setting and then every child also prints it parent.   
   * Or should it be a child only prints its parents if it is enabled for that child.   
   * Or do we automatically figure out which children have duplicate names, and automatically print their parent.   (will take more time to address)

JimLewis, 2022-01-31:
Also in the ReportAlerts/HTML tables, do we print the short name since it is listed in conjunction with its parent, or do we print the parent name with it?

SkydiverTricky, 2022-01-31:
_"Should it be a global setting and then every child also prints it parent."_
For me a global setting might work best. In a more complicated setup, I would probably just enable it globally.

_"do we print the short name since it is listed in conjunction with its parent, or do we print the parent name with it?"
I think the format in the current `ReportAlerts` is clear and makes sense.

JimLewis, 2022-02-25:
Released as part of 2022.02

closed 2022-02-25 by JimLewis