OSVVM/OSVVM #60: AlertLogPkg: ReportAlerts ignores ReportAll

davidzirbel, 2021-06-30:
I think there is a minor bug in ReportAlerts, where the hierarchy won't get printed unless the test failed even if ReportAll is true. I propose changing AlertLogPkg.vhd line 1338 from:
if (FoundReportHierVar and ReportHierarchyVar) and TestFailed then
to:
if (FoundReportHierVar and ReportHierarchyVar) and (TestFailed or ReportAll) then

Thank you!
 - David Z

JimLewis, 2021-07-01:
Hi David
Thanks for the report.   I agree with the need for the feature you are suggesting.   There are others who want this same capability.    I will make sure it gets prioritized for a near term implementation.   

That said, ReportAll currently is not a user selectable option and in the current design
was not intended to do what you are asking for.   I am not sure if the simple change
you suggest support backward compatibility or not.   However, I am committed
to making sure a feature of that sort gets added.   

Jim

JimLewis, 2021-11-10:
Hi David,
The old ReportAll parameter was ill named.   It was more like the Linux style report something only when it is signaling an error.   That was an internal only parameter that was used by ReportNonZeroAlerts.   It has been renamed to ReportWhenZero.    

A new parameter was added named ReportAll and is available in the external interface.   This one behaves the way you want it.
```vhdl
  procedure ReportAlerts ( 
    Name           : string          := OSVVM_STRING_INIT_PARM_DETECT ; 
    AlertLogID     : AlertLogIDType  := ALERTLOG_BASE_ID ; 
    ExternalErrors : AlertCountType  := (others => 0) ;
    ReportAll      : Boolean         := FALSE 
  ) ;
```

It is also available in our newer methodology with EndOfTestReports:
```vhdl
  procedure EndOfTestReports (
    ReportAll      : boolean        := FALSE ;
    ExternalErrors : AlertCountType := (0,0,0) ;
    Stop           : boolean        := FALSE
  ) ;
```

In the 2021.10 release, EndOfTestReports will subsume ReportAlerts.   I will be doing an announcement as part of the 2021.10 release to explain what has changed and why you want to do this - it has to do with the new test reporting.    

I have a couple of days needed to wrap up on 2021.10.   You will need to be using the OSVVM scripting in conjunction with EndOfTestReports.

JimLewis, 2021-11-16:
Released as part of 2021.10.

closed 2021-11-16 by JimLewis