OSVVM/OSVVM #61: Intelligent Coverage - Protected type restricted to use in different files

jocorso, 2021-07-14:
I am currently evaluating ghdl as vhdl simulation tool and OSVVM as verification methodolody tool. To carry out this evaluation I created a [gitlab](https://gitlab.com/jcorrecher/vedic_multiplier) repository with a simple design with a particular multiplier.

In the [simulation folder](https://gitlab.com/jcorrecher/vedic_multiplier/-/tree/master/sim/runs/vedic_multiplier.ghdl) you can find a detailed info to cover all the simulation steps. Following verification plan and architecture, I designed a package that contains different procedures to carry out all the functionalities to validate the design. As you can see in the picture attached on readme file, I created a modular project, where different functionalities are declared in different files.

As far as I understood, a good way to cover functionality in that design is to analize all the possible input cross values, by using intelligent coverage. As you know I have to use a protected type to generate inputs, and to cover it. This protected type can't be shared in files (driver and monitor files). So the work-around I found was declare a protected type to generate input data, and declare another protected type to cover data. It seems to work but I think is a tricky solution. 

Is there any other way to do this?

JimLewis, 2021-07-14:
@jocorso  You can declare a shared variable in a package and then it can be available to all of your design.

Shared variables can be passed through subprogram interfaces (just procedures prior to 2019, and impure functions with VHDL-2019) - they are a variable parameter that is inout.

With VHDL-2019 shared variables can be on an entity interface.   Although I think you need a patch release of current tools to have this feature working (Aldec). 

The next revision of OSVVM will have additional capability for coverage modeling - similar to alertlog that allows access to the structure from anywhere.   CoveragePkg is my current focus.

jocorso, 2021-07-16:
@JimLewis maybe I miss something... or I didn't understand well...

in [osvvm_pkg.vhd](https://gitlab.com/jcorrecher/vedic_multiplier/-/blob/master/sim/runs/vedic_multiplier.ghdl/hdl/osvvm_pkg.vhd) file I added in declaration package the following line:
```vhdl :file: osvvm_pkg.vhd
shared variable v_cross : CovPType;
```
and I commented on package body procedures, the declaration variables for cross data:

```vhdl :file: osvvm_pkg.vhd
  procedure osvvm_gen_input_rand (
  .....
    ) is
    variable v_log    : AlertLogIDType;
    variable v_a_rand : RandomPType;
    variable v_b_rand : RandomPType;
--    variable v_cross  : CovPType;
    variable v_cnt    : natural := 0;
    variable v_data_a : integer := 0;
    variable v_data_b : integer := 0;
  begin
```
```vhdl :file: osvvm_pkg.vhd
  procedure osvvm_fcoverage_cross (
  .....
    ) is
    variable v_log   : AlertLogIDType;
--    variable v_cross : CovPType;
  begin
```

In the design I am evaluating, I designed a multiplier (based on vedic mathematics) with random data with intelligent coverage. Second test results showed in that CI [job](https://gitlab.com/jcorrecher/vedic_multiplier/-/jobs/1386403451) a multiplication 8bit by 8bit with random data is done. I set 2ns as clock period, so: 2^8 * 2^8 * 2e-6 = 131.07 us is needed to evaluate it.

The final report in that job says:

```console
%% Log   INFO    in Final Report, Input Data Cross Covered: true at 131323 ns
```
However if I modified the files as I show you above, the coverage spends more time...
```console
%% Log   INFO    in Final Report, Input Data Cross Covered: true at 196509 ns
```
 I guess intelligent coverage is not applying...

JimLewis, 2021-07-16:
> I guess intelligent coverage is not applying...

The declaration in the package is not going to change your test results.   It only makes the coverage object visible to any place that references the package.    So there must be something that you changed in the two different versions of the code.

jocorso, 2021-07-17:
@JimLewis maybe I didn't explain it properly.
 I created a [branch](https://gitlab.com/jcorrecher/vedic_multiplier/-/tree/OSVVMICoverage) in order to ilustrate the modifications from [original file](https://gitlab.com/jcorrecher/vedic_multiplier/-/blob/master/sim/runs/vedic_multiplier.ghdl/hdl/osvvm_pkg.vhd) to [modified file](https://gitlab.com/jcorrecher/vedic_multiplier/-/blob/OSVVMICoverage/sim/runs/vedic_multiplier.ghdl/hdl/osvvm_pkg.vhd).

This is the [trunk simulation result](https://gitlab.com/jcorrecher/vedic_multiplier/-/jobs/1386403451).
And this is the [branch simulation result](https://gitlab.com/jcorrecher/vedic_multiplier/-/jobs/1432343712).

JimLewis, 2021-07-19:
OK.  I have cloned your repository and checked out the OSVVM branch.

In both:  `osvvm_gen_input_rand` and `osvvm_fcoverage_cross` you 
both create coverage and collect coverage (ICover).   The conditionals
in there make it hard to follow, however, I am guessing that the 
ICover in `osvvm_gen_input_rand` was only needed when they used
separate randomization objects.   

Is the value that  `osvvm_gen_input_rand` seen at the same time 
by `osvvm_fcoverage_cross` or is the value seen by `osvvm_fcoverage_cross` 
delayed by one or more clock cycles.   If it is delayed, then you will
have repeat values (but only ones that repeat within the delay
to record the values.

JimLewis, 2021-07-19:
As I reflect on your original statement, I suggested putting the coverage object in a package since you sounded worried about defining your driver and monitor coverage separately.

There is nothing wrong with having your driver and monitor coverage separate as long as when you create stimulus, you can prove that that stimulus is the actual stimulus applied.   Many of the examples I show on functional coverage depend on first verifying that when you drive particular values (or hand those values off to a procedure to be driven), that those values actually get driven - as then it is valid to collect the coverage on the driver side.

JimLewis, 2021-07-19:
WRT: 
```vhdl
  assert not (G_DATA_W >= 32) report "OSVVM doesn't allow range greater than 32" severity failure;
```
You can do larger than 32 bits by breaking the value into pieces and doing cross coverage.

```vhdl
  assert not (G_CROSS = true and G_DATA_W > 8) report
    "Enable Cross Coverage with higher widths degrades the performance simulation" severity failure;
```
Does the random one also run slow for more than 8 bits?   

With math, eventually you cannot do full enumeration tests.   For example, just trying to do incrementing values on 20x20 will take quite some time to simulate.     So instead, you need to have an algorithm that produces a pretty good set of numbers.

jocorso, 2021-07-20:
@JimLewis  Thank you to take some time to clone both projects and analyze them.
I Think I understood Intelligent coverage theory, However it looks like I didn't understand how to apply on this gitlab project.

Most of make conditionals are generics declared in the testbench to generate different test cases. 
When you execute the following command:
```console 
make GEN_DATA_TYPE=false
``` 
`osvvm_gen_input_rand` is called by using independent random data.
However, when you execute the following command:
```console 
make GEN_DATA_TYPE=false GEN_CROSS=true
``` 
`osvvm_gen_input_rand` is called as cross random data.

When I say cross random data, I mean input data generated as random data, but trying to combine all the possible values for input data.

I am using iCover  and RandCovPoint in `osvvm_gen_input_rand` when cross random data is set, in order to reach all the possible input combinations ASAP.
I created `osvvm_fcoverage_cross` just for acquire all the possible imput cross random values and notify them.

Related to the assert you are right, I will update it.

As far as random time performance is concerened, if you observe first and second test on [test_8bit](https://gitlab.com/jcorrecher/vedic_multiplier/-/pipelines/329101933) you'll realize that:
- Test 1/5 for incremental input data takes 2min and 1sec to finish the simulation.
- Test 2/5 for random input data takes 5min and 2sec to finish the simulation.

So I created the last assert in order to don't increase input width for cross random data. As you said I can create an algorithm to produces a pretty good set of numbers in order to cover the main parts of the architecture. Just for small width inputs I can check all the possible values, but for medium or big width inputs I need to create significative values to excercise the key parts of the architecture.

closed 2021-07-20 by jocorso