OSVVM/OSVVM #68: ScoreboardPkg is missing in OsvvmContext

SebDunst, 2022-03-31:
Hi Jim,
I made some tests with your new release 2022.02 and found a possible bug in the "OsvvmContext.vhd" File. There is no ScoreboardPkg included. Maybe this is your intention, but I don't think so.

Best regards,
Sebastian

JimLewis, 2022-03-31:
Hi Sebastian,
It is not intended to be part of the context.   The problem is, it is a generic package that defines types.   If we were to include both 
```
use osvvm.ScoreboardPkg_slv.all ; 
use osvvm.ScoreboardPkg_int.all ;
```
If in the test, we refer to only package, then we can reference the scoreboard using simple names for the type.  Such as:
```
use osvvm.ScoreboardPkg_slv.all ; 
. . .
signal Sb1 :  ScoreboardIdType ; 
```

OTOH, if we always include the both, then we must refer to the types as follows:
```
signal Sb1 : osvvm.ScoreboardPkg_slv.ScoreboardIdType ; 
```

As a result, if we were to add one or both, then it could potentially break people's code.

That said, did you find a release with one of them referenced?    Even though for the reasons above I am pretty sure there isn't, I did look through SmartGit to try to find something, but was not able to.

SebDunst, 2022-04-01:
You are totally right. It has never been part of the OsvvmContext.
But with release 2022.02 I tried for the first time to do my includes with the context instead of all files separately. Maybe the ScoreboardPkg_user_guide is at this point a little bit unclear.

Regards,
Sebastian

JimLewis, 2022-04-02:
@SebDunst   Goog catch.    I have updated the ScoreboardPkg_user_guide and pushed it.   Currently it is only linked through the dev branch of OsvvmLibraries, but the March update will happen shortly.

closed 2022-04-02 by JimLewis