OSVVM/OSVVM #72: Race conditions / buffer issues in console output

riedel-ferringer, 2022-04-28:
I think that there are problems with console output, esp. wrt. to vcom from ModelSim.
Steps to reproduce, it should work with any project:
- vsim -c
- source osvvm/Scripts/StartUp.tcl
- build ... (build osvvm)
- build someproject.pro
- analyze top.vhd; puts "abcdef"

You will see that "abcdef" and the output of "vcom top.vhd" are not in succession, but interleaved, like so. 
-- Loading context declaration OsvvmContext
-- Loading context dec**abcdef**
laration OsvvmCommonContext
-- Compiling entity tc_counter_ro
-- Compiling architecture rtl of tc_counter_ro
...

Remarks:
- This is not a problem in GUI mode. The output of VCOM goes to the GUI, the output from PUTS goes to the console.
- This is a problem in console / batch mode: Both outputs go to the console and somehow get mixed up.
- I think it is a buffering issue. Using "echo" instead of "puts" in the respective tcl files resolves the issue
- According to documentation, puts is buffered. I don't know about echo. 

Possible solution??
- replace all puts with echo
- however, puts and echo have different "targets": puts goes to console, while echo goes to ModelSim window
- This also affects what is written to the transcript...

JimLewis, 2022-04-28:
puts is TCL
echo is Unix and Siemens internal script stuff.
General stuff cannot use echo

In the Siemens specific stuff, echo can be used.   However, that is not a general thing.

I am having issues with creating transcripts for GHDL.   Maybe using the simulator transcript can be set aside for a more general print handler that can route all IO to a transcript file.   Maybe the print handler can make this happen in a tool
specific manner to better handle the above.

riedel-ferringer, 2022-04-28:
Yes I know. ModelSim has an internal "echo" command. And somehow, it badly interferes with puts. I think they simply forgot a 'flush' at the end of the vcom command... 

Anyway, I think your idea is very good, to use a general print handler. In ModelSim, there is another issue using transcripts (which would be solved that way): They don't work the same way in the different modes. Specifically, the 'transcript' command is not supported in batch mode - god knows why. So OSVVM's method of changing the transcript file during executions doesn't work when "vsim -batch" is used.

Maybe a lightweight solution would be to replace calls of raw puts (for stdout) to some specific file. Then there wouldn't be any more conflicts. But I understand that this has significant impacts.

riedel-ferringer, 2022-05-19:
It seems the problem was on my end after all (sorry for that)
When I call
`vsim -batch -do "build project.pro"`
the output is fine. However, when I call it with a pipeline operator to postprocess the output
`vsim -batch -do "build project.pro" | cat`
the output is somehow mixed up.

This is a buffering problem which can be solved with unbuffer:
`unbuffer -p vsim -batch -do "build project.pro" | cat`

closed 2022-05-19 by riedel-ferringer

JimLewis, 2022-05-19:
Thanks.   Can you post that to OSVVM.org form as a problem and your solution so that if other run into it they will know what to do.