OSVVM/OSVVM #73: Missing wait for 0 ns in else-branch in WaitForLevel procedure

Paebbels, 2022-05-02:
The `WaitForLevel` procedures have no else-branch with a delta-cycle delay in it.

If a `WaitForLevel` is used to also update signal assignments when the next `wait` statement is reached, it's not guaranteed by the implementation that all execution paths will cause an assignment.

Suggested solution:
```vhdl
if ... then
  -- wait until ...
else
  wait for 0 ns;
end if;
```

JimLewis, 2022-05-02:
Can you share a use model?

Paebbels, 2022-05-02:
```vhdl
dut_input <= '1';
WaitForLevel(dut_output);

dut_input <= '0';
wait until dut_error = '1';
```

This might never assign `dut_input`, because `WaitForLevel` doesn't always wait, but a user expects a wait procedure to behave like a wait statement. (That's the intension of a wait procedure, to abstract wait statements, right?)

JimLewis, 2024-11-24:
Lets take a look at something I am considering adding.

```vhdl
procedure WaitForRisingEdge(signal Clk: std_logic) is
begin
  if not Rising_Edge(Clk) then
    wait until Rising_Edge(Clk) ; 
  end if ; 
end procedure WaitForRisingEdge ; 
```

The expectation of any WaitFor is not that we are waiting, but we are waiting if that is appropriate.   In this example in particular it shows that it would be inappropriate to artificially add a wait for 0 ns in the else branch.   

Also the WaitForClock with a count value does not wait at all if the count is 0.

There are two examples that show that having a wait for 0 ns is not the right answer.   

Please re-open and comment if you disagree.

closed 2024-11-24 by JimLewis

Paebbels, 2024-11-24:
But VHDL itself always waits and updates values to signals. OSVVM is encapsulating VHDL waits into procedures for easier use, but some of these Wait procedures won't update signal values. As a result, every call to a Wait procedure of OSVVM must be followed by a wait for 0 ns to ensure signal updates.

I strongly think this is now how OSVVM's Wait procedures should be designed.

Is there any drawback in having a `wait for 0 ns;` in an else-branch?

JimLewis, 2024-11-25:
Back to my example of WaitForRisingEdge, the implication is that it is finding a rising edge.   If it does an additional wait for 0 ns when it finds it is already at a rising edge, then it will not be at a rising edge.   

At a minimum we should hold off on adding it until we see more patterns.