OSVVM/OSVVM #76: Make the AlertLogName column width of log-output configurable

riedel-ferringer, 2022-09-09:
To provide better formatted output, I think that the AlertLogName column should be customizable wrt. to it's width.
I see two (three) ways here:

- Define min width only. If the user create a longer AlertLogName, this min width automatically increases. Of course, this results in a misalignment every time a longer AlertLogName is created. 
- Define max width only. If the name is too long, shorten it and append "..." to the end
- (Define min/max and combine both methods)

Optional Enhancements:
- Allow to configure the justification (left/center/right)
- Allow the delimiter (which is currently a comma) to be configurable

JimLewis, 2022-09-12:
WRT minimum width, have you tried:
```
    SetAlertLogOptions(WriteTimeLast => FALSE) ; 
    SetAlertLogOptions(TimeJustifyAmount => 15) ; 
    SetAlertLogJustify ; 
```
After setting all of the AlertLogIDs, if you call SetAlertLogJustify, it will automatically create a field width that is big enough for printing all IDs.   

It makes you think about how long you want lower level IDs' that don't print often.

Currently it is left justified.

JimLewis, 2024-07-09:
Most of this has been implemented:  
Current defaults are time First and Justify the ID printing.   Justification is calculated on the fly.

JimLewis, 2024-07-09:
Should we consider keeping a truncate count for the ID field?   If you want this capability, re-open it after I close it.

From an OSVVM perspective, the AXI Full Manager VC has some longer child names that are planned to be shortened.

closed 2024-07-09 by JimLewis