OSVVM/OSVVM #78: Traceability between expected errors/skipped tests and issue tracking software

GlenNicholls, 2022-12-15:
It would be nice to have the ability to add traceability for expected errors between the test report and wherever the issue is tracked (e.g. Jira, GitHub, etc.). 

The way I envision this looking is that when I scroll through the report, any expected error/skipped tests would have a column with links to known issues relating to that test case. Not only this, but I would also expect that the color would be different than a normal error.

For example, we have an integration test suite that differentiates between known and unknown issues (similar to what's already available in osvvm):
[image]

An example of the linking and how this is displayed with a test:
[image]
[image]

Keep in mind that our tool uses a pattern for assigning an issue to test cases based on their name instead of directives directly in a test case. 

I find features like the above useful when there is a large backlog and issues are prioritized based on the impact of the bug and new features. I.e. some issues might stay in the backlog for a *long* time.

Another use case is with open-source dependencies. In that case, we would want to have links directly to issues we/others have opened such that we can check them periodically, or be notified in the report when an expected error unexpectedly passes.

Ideally, the linked issue wouldn't be the raw URL, but instead, look like it does above (e.g. `<issue tracker> <key>-<issue #>`). That probably complicates things when you have `company.jira.com/...`/`jira.company.com/...`, but I'm guessing it wouldn't be too bad to use a regex (or possibly allow a user to add custom patterns for their internal tracking software).

JimLewis, 2022-12-17:
How would the test case or scripts acquire the information that the test is expected to fail - until an issue is fixed?

External from the VHDL code, OSVVM tracks test cases by name.   One thing that could be done is to have a yaml file that tracked which test cases were expected to fail and the corresponding issue reports, it would be easy to add this to the OSVVM reports.   

Alternately, the test expected information could be recorded in either the scripts or the VHDL test code.   

If your test tool tracks information by test name and the test tool can be called from tcl, it is possible to look it up as the report generators are running.  To make this adaptable, a tcl procedure would be called and the procedure would be used to extract the information and return a dictionary or list to OSVVM.    OSVVM would then print the additional information in the appropriate field of the reports.   To handle adaption, the OSVVM tcl would only run this procedure if it exists - hence, the procedure would be added by the user when it is needed in their environment.

GlenNicholls, 2022-12-19:
>How would the test case or scripts acquire the information that the test is expected to fail - until an issue is fixed?

For example, you may have a test that previously passed and new changes caused it to fail. After exploring the failure, it was discovered to be minor or no impact on users. In that case, we might still release the change but create a new issue and prioritize it against other work in the backlog.

Another example is a customer reports a bug for something that isn't caught in the current test suite and we create a Jira issue. We may have limited time to explore it, so we might write a test that triggers the fault but not implement the fix yet. This is common for us when a user has a super unique use case and it exposes an issue that no other customers will experience.

Is that what you were asking for clarification about or did I misinterpret your question?

>External from the VHDL code, OSVVM tracks test cases by name. One thing that could be done is to have a yaml file that tracked which test cases were expected to fail and the corresponding issue reports, it would be easy to add this to the OSVVM reports.

>Alternately, the test expected information could be recorded in either the scripts or the VHDL test code.

For me, it feels most natural to add something to the test itself. With [Python's Pytest](https://docs.pytest.org/en/7.1.x/how-to/skipping.html), marking tests with a decorator makes this very obvious:

``` py
@pytest.mark.skip(reason="test not implemented yet per <issue #/link>")
def test_the_unknown():
    pass

@pytest.mark.xfail  # fails, but we already know that so we mark it as a known failure
def test_function():
    assert False
```

In the VHDL code, maybe something like `MarkSkip(reason="not implemented")` or `MarkXFail(reason="blah")`. I'd be fine with this metadata existing in a separate file since that makes sense to me when you need to mark many tests with metadata using a test name pattern. Personally, I would find the most value in using a subprogram in VHDL, but supporting both would satisfy more use cases.

On second thought, linking to the actual issue as I proposed isn't important to me. What I want to see in the report is a reason that a test is skipped or known to fail. Using Pytest's idea, a `reason` parameter would allow me to customize this to be whatever makes the most sense:

``` vhdl
MarkSkip(reason="Not implemented: <link>")  -- link could be full URL or key/# (e.g. VITA-123, osvvm #123, ...)
MarkXFail(reason="... reference <link>")
```

The important part here is that my test report contains this metadata. If a test is skipped or known to fail, I want to know why that is, otherwise, it can be painful to track that reason down. Adding the reason to the report makes my life easy when I can add a link to a Jira issue that contains other relevant details (e.g. related issues, priority/impact, debugging steps, etc.).

JimLewis, 2026-08-29:
Added 
```
ExpectedStatus [EXPECTED_RESULT Failures Errors Warnings "Optional Message"] and
KnownStatus [EXPECTED_RESULT "Message"]
```
Message is included in an HTML table and may contain HTML tags (such as hyperlinks).

If ExpectedStatus matches the test passes, otherwise, the test fails.  So if called on a passing tested with something other than PASSED, it will FAIL.

Examples of usage:
```
RunTest   Tb_StopCount_1.vhd
ExpectedStatus STOPLIMIT 0 0 0 "Expected Stop Limit"

RunTest   Tb_StopCount_1.vhd
KnownStatus STOPCOUNT "Checking that Stop Limit Fails - even if count is 0 and it would otherwise pass"
```

Test Output:

[image]

JimLewis, 2026-08-29:
Current implementation is on the dev branch.   If you try it out, let me know what you like and don't like.

JimLewis, 2026-08-29:
Reopened since count of untracked vs tracked errors not done

JimLewis, 2026-08-30:
Now have tracking for untracked errors and changes in tracked errors

[image]

closed 2026-08-30 by JimLewis