OSVVM/OSVVM #79: CoveragePkg: AddCross with IgnoreBin Error

SebDunst, 2023-06-22:
Hi Jim,

here is the code which shows the error (with version 2022.02):

```
architecture rtl_sim of tb_coverage is

begin
   
  p_test : process
    variable v_mycov1   : CoverageIDType;
    variable v_mycov2   : CoverageIDType;
  begin
    --SetAlertStopCount(ERROR, 1);
    v_mycov1 := NewID("MyCov1");
    v_mycov2 := NewID("MyCov2");
    wait for 0 ns;

    -- Cov1 (working without errors)
    SetMerging(v_mycov1, True);

    AddCross(v_mycov1,
             GenBin(1,5),
             IgnoreBin(7,8) & GenBin(6,10));

    WriteBin(v_mycov1);

    -- Cov2 (working but with errors)
    SetMerging(v_mycov2, True);

    AddCross(v_mycov2,
             IgnoreBin(3,4) & GenBin(1,5),
             IgnoreBin(7,8) & GenBin(6,10));

    WriteBin(v_mycov2);

    std.env.stop;
  end process;

end;
```

The difference between v_mycov1 and v_mycov2 is that there is an additional IgnoreBin. This additional IgnoreBin results in the following error:
```
%% Alert ERROR    in MyCov2, CoveragePkg.InsertBin (AddBins/AddCross): ignore bin dropped.   It is a subset of prior bin
```

The result of v_mycov2 is just fine, but to get this result the SetAlertStopCount must not be set to 1.
Is this a bug or intended?

Regards,
Sebastian

JimLewis, 2023-06-23:
Hi Sebastian,
On travel.   Hope to look at it today or tomorrow.   That said, I don't like what I am hearing.   

Best Regards,
Jim

JimLewis, 2023-06-23:
Hi Sebastian,
Looks like a bug.   I will quantify it more later.   

What follows is a work around.   Build your ignore bins first, and then add your regular bins.

It can be done with either broad set of ignore bins
``` vhdl
    AddCross(v_mycov2,IgnoreBin(3,4), ALL_COUNT) ;
    AddCross(v_mycov2,ALL_COUNT, IgnoreBin(7,8)); 
    AddCross(v_mycov2,GenBin(1,5),GenBin(6,10));
```

Or a focused set of ignore bins
``` vhdl
    AddCross(v_mycov2,IgnoreBin(3,4), GenBin(6,10,1)) ;
    AddCross(v_mycov2,GenBin(1,5,1), IgnoreBin(7,8)); 
    AddCross(v_mycov2,GenBin(1,5),GenBin(6,10));
```

JimLewis, 2023-06-24:
Hi Sebastian, 
First I improved the message so it prints which bins it did not like and your original test case prints:
```
# %% Alert  ERROR     in MyCov2,   CoveragePkg.InsertBin (AddBins/AddCross): ignore bin (3 to 4) (7)  dropped.  It is a subset of prior bin at 0 ns
# %% Alert  ERROR     in MyCov2,   CoveragePkg.InsertBin (AddBins/AddCross): ignore bin (3 to 4) (8)  dropped.  It is a subset of prior bin at 0 ns
# %% Alert  ERROR     in MyCov2,   CoveragePkg.InsertBin (AddBins/AddCross): ignore bin (3) (7 to 8)  dropped.  It is a subset of prior bin at 0 ns
# %% Alert  ERROR     in MyCov2,   CoveragePkg.InsertBin (AddBins/AddCross): ignore bin (4) (7 to 8)  dropped.  It is a subset of prior bin at 0 ns
```

These are clearly just overlapping ignore bins.    The Alert there should have been a log DEBUG, so I changed it to that.

I have pushed this to the Dev branch.    

Thanks for reporting the bug.   

Cheers,
Jim

JimLewis, 2023-06-24:
This is now fixed on the Dev branch

closed 2023-06-24 by JimLewis

SebDunst, 2023-06-26:
Hi Jim,
changing the log level from ERROR to DEBUG is certainly a good solution. But I do not understand why there is an overlapping issue at all? Where do the ignore bins overlap? (3,4) and (7,8).

Regards,
Sebastian