OSVVM/OSVVM #82: AlertLogPkg: AffirmIf with std_match (as implemented in the ScoreBoardPkg)

SebDunst, 2023-07-18:
Hi Jim,

After playing around with some value checks in my testbench (2022.02), I discovered a different behaviour when checking a value in the Scoreboard (MyFifo.Check) and checking a value with AffirmIf(MyVal=ExpVal,...).

In my case the Scoreboard is handling the "don't cares" correct by the use of std_match, but the AffirmIf check is giving me an error (ignoring the don't cares).

Do you have any advice how to deal with these two differnt behaviours?
Would it be a good idea to make a global option, where the user is able to select the AffirmIf behaviour (classic vs std_match)?

Regards,
Sebastian

JimLewis, 2023-07-18:
Hi Sebastian,
The intent is that both ScoreboardPkg_slv and AffirmIfEqual/AlertIfEqual/AlertIfNotEqual all use osvvm.AlertLogPkg.MetaMatch (which is similar to std_match - differences noted below).

For AffirmIf, it is up to you to use MetaMatch, std_match, or "=" in your expression, `AffirmIf(MetaMatch(MyVal, ExpVal), ...` or alternately use the simpler expression, `AffirmIfEqual(MyVal, ExpVal, "Simplified Message prefix");`   Note that AffirmIfEqual already prints the "Expected: ExpectedVal" and if there is an error, "/= Actual: ActualVal".

I just did a manual check of AffirmIfEqual and AlertIfEqual/AlertIfNotEqual in the current release.  I found that they all use MetaMatch with the std_ulogic family.    That said, it is possible I missed something, so it would be helpful to know which one is doing something unexpected.

If there is an issue, the work around is to use the longer `AffirmIf(MetaMatch(A, B) , ....` instead of `AffirmIfEqual`.   

With respect to 2022.02, the release notes say that:
* ScoreboardPkg_slv switched to using MetaMatch in 2020.10.
* AlertLogPkg, AffirmIfEqual switched to using MetaMatch in 2020.12.    

I don't see any notes about something being fixed later, but it could have been.

MetaMatch is like std_match except that MetaMatch also considers X = X as true where std_match considers this to be false. 
 MetaMatch is uses this table:
```vhdl
  constant MetaMatchTable : BooleanTableType := (
    --------------------------------------------------------------------------
    -- U      X      0      1      Z      W      L      H      -
    --------------------------------------------------------------------------
    (TRUE,  FALSE, FALSE, FALSE, FALSE, FALSE, FALSE, FALSE, TRUE),  -- | U |
    (FALSE, TRUE,  FALSE, FALSE, FALSE, FALSE, FALSE, FALSE, TRUE),  -- | X |
    (FALSE, FALSE, TRUE,  FALSE, FALSE, FALSE, TRUE,  FALSE, TRUE),  -- | 0 |
    (FALSE, FALSE, FALSE, TRUE,  FALSE, FALSE, FALSE, TRUE,  TRUE),  -- | 1 |
    (FALSE, FALSE, FALSE, FALSE, TRUE,  FALSE, FALSE, FALSE, TRUE),  -- | Z |
    (FALSE, FALSE, FALSE, FALSE, FALSE, TRUE,  FALSE, FALSE, TRUE),  -- | W |
    (FALSE, FALSE, TRUE,  FALSE, FALSE, FALSE, TRUE,  FALSE, TRUE),  -- | L |
    (FALSE, FALSE, FALSE, TRUE,  FALSE, FALSE, FALSE, TRUE,  TRUE),  -- | H |
    (TRUE,  TRUE,  TRUE,  TRUE,  TRUE,  TRUE,  TRUE,  TRUE,  TRUE)   -- | - |
    );
```

SebDunst, 2023-07-18:
Hi Jim,

I didn't realize that I only have to use the MetaMatch function.

Thanks a lot!

Regards,
Sebastian

closed 2023-07-18 by SebDunst