OSVVM/OSVVM #84: Add possibility to use falling clock edge as the active one.

KrzysztofZyla, 2023-08-18:
(This was discussed some time ago, via email)
Now only rising clock edge can be the active edge. In TbUtilPkg.vhd we have

><PRE>constant CLK_ACTIVE : std_logic := '1' ;</PRE>

And this value is used in procedures like WaitForTransaction, WaitForClock, CreateReset and others.
If subprograms using CLK_ACTIVE had this value passed as a parameter (CLK_ACTIVE as the default value) it would be possible to also use the falling clock edge as active.

JimLewis, 2023-08-18:
Are you thinking then pass it as an optional parameter?   That might make sense for a number of them.   I put the constant CLK_ACTIVE in there to remind me to think about how to make it more generic.   For WaitForClock, maybe there should also be a WaitForRisingClock and WaitForFallingClock?

KrzysztofZyla, 2023-08-19:
> Are you thinking then pass it as an optional parameter?

Yes. It would give the user an alternative to use them in that way without modifying OSVVM library sources.
>maybe there should also be a WaitForRisingClock and WaitForFallingClock?

Didn't thought about it, but for sure WaitForRisingClock and WaitForFallingClock is self-explanatory and more intuitive.

JimLewis, 2024-08-03:
With 2024.07 WaitForClock (and all other subprograms that detect a clock edge) now have a ClkActive optional (initialized to CLK_ACTIVE constant = '1') parameter.

```vhdl
  procedure WaitForClock ( signal Clk : in std_logic ;  constant Delay : in time ; constant ClkActive : in std_logic := CLK_ACTIVE) ;
  procedure WaitForClock ( signal Clk : in std_logic ;  constant NumberOfClocks : in integer := 1 ; constant ClkActive : in std_logic := CLK_ACTIVE) ;
  procedure WaitForClock ( signal Clk : in std_logic ;  signal Enable : in boolean ; constant ClkActive : in std_logic := CLK_ACTIVE) ;
  procedure WaitForClock ( signal Clk : in std_logic ;  signal Enable : in std_logic ; constant EnableActive : std_logic := '1' ; constant ClkActive : in std_logic := CLK_ACTIVE) ;
```

closed 2024-08-03 by JimLewis