OSVVM/OSVVM #87: ScoreboarGenericPkg: Make "match" function

SkydiverTricky, 2023-10-19:


JimLewis, 2024-07-09:
First quick test of this shows that NVC does not support impure functions as package generics.    I submitted a bug report.   I will try again when I have a version of NVC that does support it.

JimLewis, 2024-07-09:
You know the difference between commercial and open source tools.   With commercial tools you have to wait until the next update for a fix - this may be 6 months to a year.   OTOH, the issue I reported with NVC is now fixed.   :)

Testing now with current and some older releases (of commercial tools).    For commercial tools we want to support some history.   For open source tools, we expect people to update to the latest if that is necessary.

JimLewis, 2024-07-10:
On Dev branch.   Will be part of upcoming 2024.07 release

closed 2024-07-10 by JimLewis