OSVVM/OSVVM #92: [Feature] Make libraries such as NamePkg generic

nselvara, 2024-06-18:
Hey there, I really like the OSVVM library. As a feature enhancement, I'd like to suggest making the NamePkg to a generic array package. 
This would allow to be easily instantiated/overloaded with custom types.

Suggested code/package:
```vhdl
generic (
    type array_elem_t; --! Set the type of the array elements
    REALLOC_CHUNK_SIZE: natural := 2**8 --! Set something with power of 2
);

type dynamic_array_t is protected
    procedure append(value: array_elem_t);
    procedure set(value: array_elem_t; index: positive);
    impure function get(index: positive) return array_elem_t;
    impure function get_count return natural;
    impure function remove(index: positive) return integer; -- 0 = OK, -1 = failure
    impure function is_set return boolean;
    procedure clear_array;
    procedure deallocate_array;
end protected;
```

JimLewis, 2024-07-09:
A general set/get package is in the plans.   Unfortunately it cannot be used in OSVVM core as we have at least one simulator that is lagging in support for generic packages.

nselvara, 2024-07-09:
I seee. That would be really cool. I've modified one of your `message_pkg` (I think, it's called something like that) to suit the need.
Which simulator is currently lacking the support?

JimLewis, 2024-09-29:
This also requires usage of 2019 sense of generic types as then we will know that it is an array type and can do interesting things like append or prepend to the array.

That said there are simulators implementing these type of features and it can be introduced as a 2019 only feature.

nselvara, 2024-10-10:
I see, yeah this makes kind of sense. I'm just not sure if maintaining duplicate code, makes sense at all, but on the other hand adding one more duplicate code is also not a huge deal.

JimLewis, 2025-02-13:
I am looking forward to better 2019 support.   Be sure to encourage your vendors and be specific about what you want - here using unspecified types as a formal generic.