OSVVM/OSVVM #93: Using Named Association with AlertIf Gives "Ambiguous Subprogram" Error

SkydiverTricky, 2024-07-31:
I am using 2024.5, but the same error exists in Dev. If you used named association for 

```vhdl
AlertIf(AlertLogID => id, 
        condition   => (a > b),
        Message     => "signature clash" 
        Level     => ERROR );
```

This is declared on [L178](https://github.com/OSVVM/OSVVM/blob/c9ec3ffc79a6e9520dd13305f4b68594b44752cb/AlertLogPkg.vhd#L178) of AlertLogPkg

This clashes with the signature on line [L671](https://github.com/OSVVM/OSVVM/blob/c9ec3ffc79a6e9520dd13305f4b68594b44752cb/AlertLogPkg.vhd#L671) which is listed in the "deprecated" section.

JimLewis, 2024-08-01:
I did a push on the Dev branch to address this.   I think all I needed to do is the change the name of the first parameter of the first parameter.

JimLewis, 2024-08-03:
If you get a chance to try the update on Dev, let me know.

JimLewis, 2024-09-29:
Update that fixes this released with 2024.09.   Please re-open if you still have issues

closed 2024-09-29 by JimLewis