OSVVM/OSVVM #95: Issue facing of compilation AlertLogPkg.vhd using GHDL

adeel-akhter99, 2024-09-29:
ghdl -a --std=08 --work=osvvm OSVVM/AlertLogPkg.vhd 
OSVVM/AlertLogPkg.vhd:115:1:warning: package "alertlogpkg" was also defined in file "OsvvmLibraries/osvvm/AlertLogPkg.vhd" [-Wlibrary]
OSVVM/AlertLogPkg.vhd:726:59:error: conversion allowed only between closely related types
OSVVM/AlertLogPkg.vhd:1638:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:1757:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:1793:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:1929:46:warning: declaration of "alertcount" hides variable "alertcount" [-Whide]
OSVVM/AlertLogPkg.vhd:1935:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2008:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2043:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2109:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2143:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2215:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2258:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2350:7:warning: declaration of "alertcount" hides variable "alertcount" [-Whide]
OSVVM/AlertLogPkg.vhd:2358:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2551:7:warning: declaration of "alertcount" hides variable "alertcount" [-Whide]
OSVVM/AlertLogPkg.vhd:2556:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2645:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2717:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2815:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:2826:16:warning: declaration of "alertcount" hides variable "alertcount" [-Whide]
OSVVM/AlertLogPkg.vhd:3888:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:4060:16:warning: declaration of "buf" hides variable "buf" [-Whide]
OSVVM/AlertLogPkg.vhd:4611:96:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:4735:102:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:4860:97:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:4985:103:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5471:62:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5472:43:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5605:62:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5606:43:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5738:62:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5739:44:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5871:62:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5872:44:error: no overloaded function found matching 'to_string'

JimLewis, 2024-09-29:
What release of OSVVM are you using?   Are you compiling OSVVM with the provided scripts?
Which version of GHDL are you using?

The following message from above seems to say that you already compiled OsvvmLibraries/osvvm/AlertLogPkg.vhd and that you are compiling another source of the file.   
> OSVVM/AlertLogPkg.vhd:115:1:warning: package "alertlogpkg" was also defined in file "OsvvmLibraries/osvvm/AlertLogPkg.vhd" [-Wlibrary]

When OSVVM compiles the library, it uses::    
ghdl -a --std=08 -Wno-library -Wno-hide --work=osvvm --workdir=C:/tools/sim/ghdl/VHDL_LIBS/GHDL-5.0.0-dev/osvvm/v08 -PC:/tools/sim/ghdl/VHDL_LIBS/GHDL-5.0.0-dev C:/osvvm_releases/OsvvmLibraries/osvvm/AlertLogPkg.vhd

Those settings will eliminate the warnings about [-Wlibrary] and [-Whide] since these are ok with the language, but not so much with GHDL.   

The errors still have to be addressed.   These are of the form:
OSVVM/AlertLogPkg.vhd:4611:96:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:4735:102:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:4860:97:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:4985:103:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5471:62:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5472:43:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5605:62:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5606:43:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5738:62:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5739:44:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5871:62:error: no overloaded function found matching 'to_string'
OSVVM/AlertLogPkg.vhd:5872:44:error: no overloaded function found matching 'to_string'

The line numbers do not quite correlate with the current release of OSVVM, however, if I make a guess, it is referring to the to_string function with the signature:   to_string[integer_vector return string].    

This function is defined in VHDL-2019, but not VHDL-2008.   In OSVVM, for VHDL-2008, you need to compile the package deprecated/LanguageSupport2019Pkg_c.vhd to get this functionality.    I think this file was moved from the osvvm directory to the osvvm/deprecated directory either in 2024.09 or 2024.07 release.  

If you want to compile OSVVM without using the OSVVM compile scripts, you need to read OsvvmLibraries/osvvm/CHANGELOG.md to get the correct files to compile and the order in which to compile them.

JimLewis, 2024-11-24:
After following the above suggestions, if you still have issues, please re-open this.

closed 2024-11-24 by JimLewis