OSVVM/OSVVM #97: Add InitSeed Overload to DelayCoveragePkg

SkydiverTricky, 2025-01-15:
The coverage package has and `InitSeed` function for all coverage models - I think it would be useful to have an overload in the DelayCoveragePkg. Currently, all 3 IDs in the DelayCoverageIdType have to be seeded separetely.

JimLewis, 2025-01-15:
Are you looking to set the seeds to something particular or just set them all to something different?

The call to NewID calls InitSeed and gives each coverage model a unique seed.   

Otherwise what do you want InitSeed to do for the 3 IDs?

SkydiverTricky, 2025-01-15:
I like to control all of the seeding in the TB - which usually comes from an external source - usually a generic, so that it can be run in CI with a different seed on each run to catch corner cases over time. I think for these it would be nice just to have a wrapper function that sets all 3 to the same seed as I dont think it will affect much. If anyone needs all 3 seeded differently, they can just manually set each one.

JimLewis, 2025-01-15:
Have you seen SetRandomSalt?    It is intended for that application.

In your top-level testbench, do:
```vhdl
entity TestHarness is
  generic (DATE_STRING : string := "") ;
end entity TestHarness ; 
architecture structural of TbMemIO is 
  constant SaltIsSet : boolean := SetRandomSalt(DATE_STRING) ;
  . . . 
  ```

JimLewis, 2025-01-15:
I should note, I am ok with having a wrapper function.   The question is, what are its parameters:   Does it accept either 3 string or 3 integer values?

SkydiverTricky, 2025-02-14:
Probably best to offer 3x string  and  3x integer, to be consistent with all other initSeed options