OSVVM/OSVVM #99: RandReal generates incorrect values in GHDL

weilanad, 2025-02-07:
**Description**
The function `RandReal`  from the `RandomPkg` calculates falsely in GHDL.
Calculating a random real in the range of 0.499 to 0.501 results in a large number (-8.75e307).

I ran the same script on Riviera-PRO 2024.10 where this value is calculated correctly

**Expected behavior**
The value should be in the specified range of 0.499 to 0.501.  
Here the output from Riviera-PRO:
```
[...]
NOTE   : passed: 0.500027
[...]
```

**How to reproduce?**
The example folder has the following structure:
```
temp/test_RandReal.vhdl               <- example script
temp/lib/osvvm                        <- includes RandomPkg
temp/lib/OSVVM-Scripts
```

```vhd :file: test_RandReal.vhd
library IEEE;
use     IEEE.std_logic_1164.all;

library OSVVM;
use     OSVVM.RandomPkg.all;

entity test_RandReal is
end entity;

architecture test of test_RandReal is
	signal output : real := 0.0;
begin
	proc: process
		variable RV: RandomPType;
	begin
		RV.InitSeed(RV'instance_name);
		output <= RV.RandReal(0.499, 0.501);
		wait for 10 ns;

		std.env.finish;
	end process;

	assert (output >= 0.0) report "failed: " & to_string(output) severity failure;
end architecture;
```

> NOTE: Before executing the comments below the library OSVVM has to be built (for this run [commit (28327a97)](https://github.com/OSVVM/OSVVM/commit/28327a97a6b8883afb3ccf375ca03b4d2d657484) has been used).
```sh :image: ghdl/ghdl:bullseye-mcode
source lib/OSVVM-Scripts/StartUp.tcl
build lib/osvvm/osvvm.pro
RunTest test_RandReal.vhdl
```

The assertion error is shown below:
```
% RunTest Test_RandReal.vhdl
RunTest Test_RandReal.vhdl
TestName Test_RandReal
analyze Test_RandReal.vhdl
ghdl -a --std=08 -Wno-library -Wno-hide --work=DefaultLib --workdir=C:/temp/VHDL_LIBS/GHDL-5.0.0-dev/defaultlib/v08 -PC:/temp/VHDL_LIBS/GHDL-5.0.0-dev Test_RandReal.vhdl

Simulation Start time 14:29:54
simulate Test_RandReal
ghdl --elab-run --std=08 --syn-binding --work=DefaultLib --workdir=C:/temp/VHDL_LIBS/GHDL-5.0.0-dev/defaultlib/v08 -PC:/temp/VHDL_LIBS/GHDL-5.0.0-dev Test_RandReal
Error: Test_RandReal.vhdl:23:9:@0ms:(assertion failure): failed: -8.745584833876868e307
ghdl:error: assertion failed
ghdl:error: simulation failed
Error: child process exited abnormally
Simulation Finish time 14:29:55, Elapsed time: 0:00:01
SimulateError: See messages above in "simulate Test_RandReal"
For tcl errorInfo, puts $::osvvm::SimulateErrorInfo
```

A related report has been created for GHDL: 

**Context**
Please, provide the following information:

- OS: Windows + MSYS2/UCRT64 - mcode
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `GHDL 5.0.0-dev (4.1.0.r63.g9bcaa92a2)`

/cc @Paebbels

weilanad, 2025-02-07:
I was able to update my GHDL to the [latest nightly](https://github.com/ghdl/ghdl/commit/c74934214c1d9ad4ef9deaf1b2fb19de007379ef). With this version the error no longer occurs.

JimLewis, 2025-02-07:
@weilanad   You can just tag me in your future GHDL bug reports if you think it could be OSVVM.

closed 2025-02-07 by JimLewis