OSVVM/OSVVM #100: Scoreboard aliasing empty to IsEmpty can overload value names

GCHQDeveloper147, 2025-02-10:
An import of ScoreboardPkg_slv resulted in a simulation error whereby an element of a record constant (called empty) was replaced with a call to IsEmpty, which was not a part of that record. In our specific case, we received an error: `(vcom-1065) Element "IsEmpty" is not defined for record type "lib.interface.stream".`

As `empty` is used in a lot of standard signal definitions (especially packet interfaces), assuming this is intended behaviour in VHDL, aliasing empty to IsEmpty can cause breaking issues.

Would it be possible to workaround this, either by removing the alias or making the naming more scoreboard-specific?

GCHQDeveloper147, 2025-02-10:
As a clarification: this only occurs when importing a package with ALL (such as `USE osvvm.ScoreboardPkg_slv.ALL;`). Importing fields directly in the code (e.g. `VARIABLE x : osvvm.ScoreboardPkg_slv.ScoreboardPType`) works.

JimLewis, 2025-02-10:
Without an example, it looks like your simulator is having issues with overload resolution.   We probably also need the version of the simulator to help with investigation.

GCHQDeveloper147, 2025-02-12:
We're running this in Questa 2025.1 (and previously had the same outcome in 2024.3).

```
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

LIBRARY osvvm;
USE osvvm.ScoreboardPkg_slv.ALL;

ENTITY test IS
END ENTITY;

ARCHITECTURE t OF test IS
  TYPE mything IS RECORD
    empty : std_ulogic_vector(1 DOWNTO 0);
  END RECORD;
  SIGNAL thing : mything;
BEGIN

  run : PROCESS
  BEGIN
    thing <= (
      empty => (OTHERS => 'H')
    );
    -- Element "IsEmpty" is not defined for record type "mything".
    WAIT;
  END PROCESS;

END ARCHITECTURE;
```

JimLewis, 2025-02-12:
Thanks for the test case.   I can confirm it fails in my version of Questa too.   That is really unfortunate.

What I see in ScoreboardGenericPkg is:
```vhdl
  -- Empty - check to see if scoreboard is empty
  impure function IsEmpty (
    constant ID     : in  ScoreboardIDType
  ) return boolean ;
  -- Tagged
  impure function IsEmpty (
    constant ID     : in  ScoreboardIDType ;
    constant Tag    : in  string
  ) return boolean ;                    -- Simple, Tagged

  alias Empty           is IsEmpty [ScoreboardIDType return boolean] ;
  alias Empty           is IsEmpty [ScoreboardIDType, string return boolean] ;
```

There is no call to empty that is aliased to IsEmpty that can be called with no parameters - as the error in Questa seems to indicate.    

I am not able to reproduce the Questa issue in NVC, GHDL, and RivieraPRO.   So I can only conclude that this is a Questa bug.

Have you reported it as a Questa Bug?

JimLewis, 2025-02-12:
OSVVM Dev branch now has a work around for the Questa Bug.
-----------------------------------------------------------------------
First I removed the alias and tested it.   Indeed it does work without the alias.   I made sure all OSVVM VC are updated to use IsEmpty rather than Empty - good to remove technical debt since Empty is soft deprecated.   

Since some will still need Empty to work, I reintroduced Empty as a function that is identical to IsEmpty rather than an alias.  That works without an issue in Questa.    This falls in line with OSVVM's policy of working around tool issues where possible - and also maintaining backward compatibility.   I have pushed these updates to the dev branch of OSVVM.   

I have now tested it also with Questa and it is no longer an issue on my version of Questa.

Be sure to submit a bug report against Questa.

Thanks for submitting a test case with the issue.

closed 2025-02-12 by JimLewis