OSVVM/OSVVM #104: Finish adding `to_string [integer_vector]`

rhinton, 2025-07-07:
I'm stuck using Questa, whose marketing manager told me a few years ago that "all our customers are happy with our VHDL-2008 offering."  Meaning they have no plan to add VHDL-2019 support.

So `AlertLogPkg` won't compile without the stubbed out `to_string` whose declaration in `LanguageSupport2019Pkg.vhd` is commented out.  Here is a textbook-style example of how I might approach implementing it.  (Compiles, but not tested.  Sorry, I can't provide a pull request.)  Feel free to copy / adapt / etc.
```vhdl
  function to_string_recursive (AA : integer_vector) return string is
    constant AA_up : integer_vector(0 to AA'length - 1) := AA;
  begin
    if 0 = AA'length then
      return "";
    elsif 1 = AA'length then
      return to_string(AA_up(0));
    else
      return to_string(AA_up(0)) & "," & to_string_recursive(AA_up(1 to AA_up'high));
    end if;
  end function to_string_recursive;
  
  function to_string ( A : integer_vector) return string is
  begin
    return "(" & to_string_recursive(A) & ")";
  end function to_string;
```

Paebbels, 2025-07-08:
@rhinton at last weeks FPGA Conference Europe, I saw the list of implemented VHDL-2019 features and we discussed what to implement next in QuestaSim :).

JimLewis, 2025-07-18:
Closing as 2025.04 has the VHDL-2019 off for QuestaSim and 2025.06 will also have it off.

closed 2025-07-18 by JimLewis